wasp-os/wasp-os #91: Init hardware FAILED on pinetime after installing

StephanStanisic, 2020-10-26:
Hi,

I just recieved my pinetime devkit a few days ago and I was trying to install wasp-os on it. After installation the screen shows "Init hardware FAILED", suggesting something went wrong in the [setup of watch.py](https://github.com/daniel-thompson/wasp-os/blob/master/wasp/boards/pinetime/watch.py.in#L87-L97). 

I am also unable to connect to the console while in the bootloader using `tools/wasptool --device DB:D8:D8:D4:5A:31 --console`, I get a "OSError: [Errno 5] Input/output error" error running both as root and normal user. ([full log](https://paste.linux.pizza/?569f515ecb11ecd9#emLrnkAJCmdGtzvK8YU3VzqXFxZBj9bJXBRNH6q1kkU))
Connecting using pynus as suggested in #23 also yields an error, this time a NoReply error. ([full log](https://paste.linux.pizza/?34b127a171acc529#FJeWEXzt4EecPViD8kfV237Y7prqTn63qJ558ZiVkpfC))

I followed the steps on the docs, first building wasp-os, flashing bootloader.hex to 0x0 with pinetime-updater (I have a ST-Link V2), and running a dfu update using the supplied ota-dfu tool. (upload did take 8 minutes, but as mentioned in a video this is to be expected)

I am able to run infinitime on the watch, and I think this may be a problem with the heart rate sensor? I am unsure if the heart rate sensor is working since it doesn't show anything useful (looks like a placeholder) using infinitime.

Any suggestions on how to debug this further are welcome!

SteveAmor, 2020-10-26:
Hi @StephanStanisic rather than building your own files, could you try the release here:
https://github.com/daniel-thompson/wasp-os/releases/tag/v0.2
It's not got all the features, or error messages, but it would be good if you could test it.

daniel-thompson, 2020-10-26:
Hi @StephanStanisic 

Hmnn... I don't recall anyone ever reporting anything like this before.

Before trying 0.2 could we have a go a trying to instrument watch.py.in to find out what is going wrong? Basically you'll see that there are six bits of hardware being initialized (battery, i2c, accelerometer, hrs, touch, vibration motor). Could you add appropriate boot_msg() calls between each one so we can get a clue about which component it is that isn't coming up.

If its the heart rate sensor it would be good to know to confirm this and, if so, I think we might be able to defer the h/ware initialization to a little later in the startup (when the error handling can be a tiny bit more graceful).

daniel-thompson, 2020-10-26:
Actually if it is the heart rater sensor then you could try this change to hrs3300.py which will defer initialization (untested but hopefully it will give you the idea). It should make the thing boot and you'll get crash report on the screen whenever you switch to the heart rate app (which will be annoying for you but I'd be interested to see the crash report in order to step towards a proper fix):

~~~
diff --git a/wasp/drivers/hrs3300.py b/wasp/drivers/hrs3300.py
index 3843887..83f716b 100644
--- a/wasp/drivers/hrs3300.py
+++ b/wasp/drivers/hrs3300.py
@@ -27,6 +27,7 @@ class HRS3300:
     def __init__(self, i2c):
         self._i2c = i2c
 
+    def init(self):
         w = self.write_reg
 
         # HRS disabled, 12.5 ms wait time between cycles, (partly) 20mA drive
@@ -50,6 +51,8 @@ class HRS3300:
         self._i2c.writeto_mem(_I2CADDR, addr, bytes((val,)))
 
     def enable(self):
+        self.init()
+
         enable = self.read_reg(_ENABLE)
         enable |= _ENABLE_HEN
         self.write_reg(_ENABLE, enable)

~~~

StephanStanisic, 2020-10-31:
So, I've tried adding the extra logging messages, and while opening up my watch to plug in my programmer again I noticed that the heart rate sensor was plugged in at an angle. So I unplugged + plugged it in again and that fixed the issue! 

I'm not sure why I didn't check the connection earlier, it must have been loosened by me folding the battery back.

I do think that adding the extra display lines could help other people with debugging, since it directly shows what's wrong instead of just "hardware failed".

closed 2020-10-31 by StephanStanisic