wasp-os/wasp-os #92: DaFlasher: "Error DFU file not found" (Pinetime)

LalaDK, 2020-10-27:
Hi, 

I'm following the Installation Guide and trying to install wasp-os on Pinetime Devkit. and I've managed to build wasp-os from source without problem, but I got  stuck doing the installation of the bootloader, particularly this step: 

`Click Select file and choose DaFitBootloader23Hacked.bin, then wait for the payload to be transferred and for the install process to complete on the watch (leaving three coloured squares on the display).`

When I do so I get the message: "Error DFU FILE NOT FOUND":

[image]

My android device is a Samsung S10 Plus (SM-G975F) with Android 10

Am I doing something wrong? 

Thanks in advance

SteveAmor, 2020-10-27:
Hi @LalaDK it looks like you have a new PineTime with Infinitime installed. Unfortunately, the DaFlasher method will not work with this new Infinitime firmware.  I don't believe there is a way to flash wasp-os OTA with Infinitime yet, you will have to use the SWD method.
https://forum.pine64.org/showthread.php?tid=9017&pid=80969#pid80969

LalaDK, 2020-10-28:
Thank you for your response!
I'll have a go at the SWD method when I acquire a STLink adapter. 
Thank you!

closed 2020-10-28 by LalaDK

LalaDK, 2020-10-31:
I got my ST-link, followed the guide and now WASP OS is up and running! 
Looking forward to tinker with it all :-) 
Thank you!