wasp-os/wasp-os #94: Freeing up space on files

yeyeto2788, 2020-10-30:
Hey @daniel-thompson,

I'm waiting for my PineTime to arrive and test this. In the meanwhile, I've been taking a look at the whole repo and this issue is more of a suggestion than an issue itself.

I have [played with MicroPython](https://github.com/yeyeto2788/MicroPythonScripts) and memory was one of my main issues when writing big applications with descriptive docstrings and 4 spaces indexing. That is why I develop this [tool](https://github.com/yeyeto2788/MicroPythonScripts/blob/master/tools/uncommend_and_upload.py) which might help you in having more space (some KB) on the boards.

Here you have a simpler implementation of it:

```python
from pyminifier import minification
from pyminifier import token_utils

class PyminiferOptions:
    """
    Abstract class to define the options used for the pyminifier module.
    """

    def __init__(self):
        """Simple constructor method to define the properties for .
        """
        # Use tabs instead of spaces
        self.tabs = False
        # Obfuscate all code
        self.obfuscate = False
        self.obf_classes = False
        self.obf_functions = False

        self.nominify = False
        self.obf_variables = False
        self.obf_builtins = False
        self.obf_import_methods = False
        self.replacement_length = 1
        self.use_nonlatin = False

_file = "./get_emojis.py"
# Open the file and read it's content.
source = open(_file).read()
# Get tokens from file.
tokens = token_utils.listified_tokenizer(source)
# Minify the file content based on the tokens
minified = minification.minify(tokens, PyminiferOptions())
# Recompute tokens from minified version.
tokens = token_utils.listified_tokenizer(minified)
# Final result on file minified.
result = token_utils.untokenize(tokens)

# Print it to the screen or save it to a file
print(result)
```

Which will convert a file like this (4004 characters):
```python
#!/usr/bin/env python3
import concurrent
import os
import time
from concurrent.futures import ThreadPoolExecutor
from urllib.parse import urlparse

import requests

proxies = {
    "http": "http://localhost:80",
    "https": "http://localhost:80",
}

def get_unicode_name(url: str):
    """Get unicode from url.
    
    Args:
        url: 

    Returns:
        String with unicode found on the URL.
    """
    url_structure = urlparse(url)
    unicode_name = os.path.splitext(os.path.basename(url_structure.path))[0]
    return unicode_name

def get_emojis():
    """Get them all from Github API.
    
    Returns:
        Dictionary with all emojis on the response
    """
    response = requests.get('https://api.github.com/emojis', proxies=proxies)
    emojis_found = {}

    if response.status_code == 200:
        json_data = response.json()

        for emoji_name, emoji_url in json_data.items():
            emojis_found[emoji_name] = {
                "url": emoji_url,
                "representation": f":{emoji_name}:",
                "unicode": f"{get_unicode_name(emoji_url)}"
            }
    return emojis_found

def create_readme(emojis: dict):
    """Create simple README.md file on CWD
    
    Args:
        emojis: Dictionary with all emojis gotten on request.

    Returns:

    """
    with open("emoji_readme.md", "w") as readme_file:
        file_data = """
| **Readme representation** | **Emoji** | **Unicode** |
| :-------------: | :-------------: | :----------: |
"""
        for emoji_name, emoji_values in emojis.items():
            file_data += f"| `{emojis[emoji_name]['representation']}` "
            file_data += f"| {emojis[emoji_name]['representation']} "
            file_data += f"| `{emojis[emoji_name]['unicode']}` |\n"

        readme_file.write(file_data)
        print("Readme file created.")

def download_file(url: str, path: str):
    """
    
    Args:
        url: URL to retrieve file from.
        path: Path of image to be created. 

    Returns:
        True if finishes correctly otherwise false.
    """
    try:
        with open(path, "wb") as emoji_image:
            response = requests.get(
                url,
                allow_redirects=True,
                proxies=proxies
            )
            emoji_image.write(response.content)
            print(f"Emoji downloaded {path}")

        return True

    except Exception:
        return False

def download_all_emojis(emojis: dict, workers: int = 100):
    """Get them all
    
    Args:
        emojis: Dictionary with all emojis gotten on request.
        workers: Amount of worker's concurrency.

    Returns:
        None
    """
    emojis_dir_path = os.path.join(os.getcwd(), 'emojis')
    if not os.path.exists(emojis_dir_path):
        os.makedirs(emojis_dir_path, exist_ok=True)

    emojis_urls = {
        emojis.get(emoji_name).get('url'):
            os.path.join(emojis_dir_path, f"{emoji_name}.png")
        for emoji_name, emoji_values in emojis.items()
    }

    with concurrent.futures.ThreadPoolExecutor(
            max_workers=workers) as executor:
        future_to_url = {
            executor.submit(download_file, url, emoji_filename): url for
            url, emoji_filename in
            emojis_urls.items()
        }

        for future in concurrent.futures.as_completed(future_to_url):
            url = future_to_url[future]

            try:
                future.result()
            except Exception as exc:
                print('%r generated an exception: %s' % (url, exc))

def main():
    start_time = time.time()
    emojis = get_emojis()
    create_readme(emojis)
    readme_time = time.time()
    download_all_emojis(emojis)
    download_time = time.time()

    message = f"""
All execution took {(time.time() - start_time)} seconds.
- Readme file: {(readme_time - start_time)} seconds.
- Download emojis: {(download_time - readme_time)} seconds.
"""
    print(message)

if __name__ == "__main__":
    main()

```

To something like this **(2688 characters)**:
```python
#!/usr/bin/env python3
import concurrent
import os
import time
from concurrent.futures import ThreadPoolExecutor
from urllib.parse import urlparse
import requests
proxies={"http":"http://local-zscaler.boehringer.com:80","https":"http://local-zscaler.boehringer.com:80",}
def get_unicode_name(url:str):
 url_structure=urlparse(url)
 unicode_name=os.path.splitext(os.path.basename(url_structure.path))[0]
 return unicode_name
def get_emojis():
 response=requests.get('https://api.github.com/emojis',proxies=proxies)
 emojis_found={}
 if response.status_code==200:
  json_data=response.json()
  for emoji_name,emoji_url in json_data.items():
   emojis_found[emoji_name]={"url":emoji_url,"representation":f":{emoji_name}:","unicode":f"{get_unicode_name(emoji_url)}"}
 return emojis_found
def create_readme(emojis:dict):
 with open("emoji_readme.md","w")as readme_file:
  file_data="""
| **Readme representation** | **Emoji** | **Unicode** |
| :-------------: | :-------------: | :----------: |
"""  
  for emoji_name,emoji_values in emojis.items():
   file_data+=f"| `{emojis[emoji_name]['representation']}` "
   file_data+=f"| {emojis[emoji_name]['representation']} "
   file_data+=f"| `{emojis[emoji_name]['unicode']}` |\n"
  readme_file.write(file_data)
  print("Readme file created.")
def download_file(url:str,path:str):
 try:
  with open(path,"wb")as emoji_image:
   response=requests.get(url,allow_redirects=True,proxies=proxies)
   emoji_image.write(response.content)
   print(f"Emoji downloaded {path}")
  return True
 except Exception:
  return False
def download_all_emojis(emojis:dict,workers:int=100):
 emojis_dir_path=os.path.join(os.getcwd(),'emojis')
 if not os.path.exists(emojis_dir_path):
  os.makedirs(emojis_dir_path,exist_ok=True)
 emojis_urls={emojis.get(emoji_name).get('url'):os.path.join(emojis_dir_path,f"{emoji_name}.png")for emoji_name,emoji_values in emojis.items()}
 with concurrent.futures.ThreadPoolExecutor(max_workers=workers)as executor:
  future_to_url={executor.submit(download_file,url,emoji_filename):url for url,emoji_filename in emojis_urls.items()}
  for future in concurrent.futures.as_completed(future_to_url):
   url=future_to_url[future]
   try:
    future.result()
   except Exception as exc:
    print('%r generated an exception: %s'%(url,exc))
def main():
 start_time=time.time()
 emojis=get_emojis()
 create_readme(emojis)
 readme_time=time.time()
 download_all_emojis(emojis)
 download_time=time.time()
 message=f"""
All execution took {(time.time() - start_time)} seconds.
- Readme file: {(readme_time - start_time)} seconds.
- Download emojis: {(download_time - readme_time)} seconds.
""" 
 print(message)
if __name__=="__main__":
 main()
```

I used this module:
```shell
$ pip show pyminifier
Name: pyminifier
Version: 2.1
Summary: Python code minifier, obfuscator, and compressor
Home-page: https://github.com/liftoff/pyminifier
Author: Dan McDougall <daniel.mcdougall@liftoffsoftware.com>
```

P.D: Hope it helps you.

daniel-thompson, 2020-10-30:
Thanks for the comment.

Perhaps I should explain the existing strategies wasp-os has w.r.t. running large programs.

The first strategy is the [dirt simple minifier that is built into wasptool](https://github.com/daniel-thompson/wasp-os/blob/37f8a47220337d8f9cd1230703200c6e6ea7a515/tools/wasptool#L70-L88). This is pretty dumb (it only strips out docstrings, comments and blank lines) but it is enough to make a big difference to RAM usage.

The second option is to transfer code that has already been byte-compiled. Byte code naturally strips out comments (the uPy compiler also strips out docstrings). I think it can also eliminate local variable names. Finally we save additional RAM because we don't have to parse the code (so no parse state variables are needed).

A fully fledged minifier would sit somewhere in the middle. It would be better than the existing minifier but is likely to be inferior to pre-compiling the byte code.

yeyeto2788, 2020-10-30:
Thanks for the quick reply, with pre-compile byte code this cannot compete against 😢 . But I guess it would be better than the [`paste`](https://github.com/daniel-thompson/wasp-os/blob/master/tools/wasptool#L65-L108) function you have implemented on the tool which is mainly what I was doing on my playground but I got weird results if I messed up on docstring with quotes. 

The script can be tweaked to get different outputs.

daniel-thompson, 2020-11-08:
I've dropped this idea into the TODO list (although ignore the fact it is targetting 0.4... I haven't yet split the TODO list into the actively planned features versus the wishlist): https://github.com/daniel-thompson/wasp-os/blob/master/TODO.rst#wasptool

I think this is likely to end up on the wish list and we'll see if somebody shows up who wants to implement it!

closed 2020-11-08 by daniel-thompson

yeyeto2788, 2020-11-09:
@daniel-thompson no problem. I think I can handle the implementation if you don't mind. I can base the implementation on the current code in `master`.

daniel-thompson, 2020-11-09:
> no problem. I think I can handle the implementation if you don't mind.

No, I don't mind. New contributors and sane pull requests both make me happy!