wasp-os/wasp-os #99: Haptic alarm feature in waspos

Izaic, 2020-11-10:
With all the features being added to waspos, it would be interesting to see a haptic "alarm" that vibrates till you tap on the screen to confirm that you've awoken.

tht-jxny, 2020-11-13:
I could try to implement this. @Izaic, how would you like to set the time / a timer? @daniel-thompson do you want to assign tasks?

daniel-thompson, 2020-11-14:
For completely unrelated reasons I've added an alarm system to the system manager (wasp.system). Currently it is being used to queue a callback that resets the step counter at midnight. However I tihnk it could also be used by an Alarm app to get a callback to turn the vibrator on (and to queue another callback to turn it off again).

Take a look at https://github.com/daniel-thompson/wasp-os/commit/d0a99d5636e9e597a44d872b5350c3588b032439 (looks for set_alarm() and cancel_alarm() ). If you grab the latest master branch you can see it in use in the step counter applications. Finally don't forget to update the submodules. I had to fix a but in micropython to get the code working.

tht-jxny, 2020-11-15:
Ok, I'll start to work on this, but I'm afraid that I probably won't finish it before the end of this month. So if anybody comes along here with more time to spare he is welcome to do so ;)

Siroj42, 2020-11-16:
I am also working on this, and my version should be almost ready. I just have to add some proper documentation and probably clean up a bit. My app currently only supports one alarm at a time. There is also a radio button to disable it completely. Here is a screenshot from the simulator.
[image]

Izaic, 2020-11-17:
Neat @Siroj42 !

@tht-jxny Sorry, i never got a notification that there was a reply to this ticket. I figured it didn't have to be anything special, just set a time, tap confirm, and then when that time comes along it starts vibrating even if the screen is off, till you tap the screen or something to stop the alarm.

daniel-thompson, 2020-11-30:
I've merged the alarm app from @Siroj42 over the weekend. It is disabled by default but is included in the flash so it just needs two lines adding to main.py to activate. Copy wasp/main.py to main.py and then add the folliowing to your copy:

~~~
from apps.alarm import AlarmApp
wasp.system.register(AlarmApp())
~~~

After that you can use `wasptool --upload main.py` to activate the app.

closed 2020-11-30 by daniel-thompson