wasp-os/wasp-os #100: Colmi P8 StepCounterApp accelerometer driver crash

Gozel, 2020-11-13:
Hi Daniel, thanks for this open source project! 
I tried to install the steps.py on the Colmi P8 and I get following output after it's uploaded to the watch:
~~~
Traceback (most recent call last):
  File "<stdin>", in <module>
  File "<stdin>", in __init__
  File "drivers/bma421.py", in reset
ValueError: invalid sensor
~~~
Can you advise?

SteveAmor, 2020-11-13:
Hi @Gozel check out #69 
It's likely that you have a newer P8 with a different accelerometer.

daniel-thompson, 2020-11-13:
Assuming you bought the new model P8 (with the magnetic charger) then this is because the sensor is not supported:
https://wasp-os.readthedocs.io/en/latest/install.html#colmi-p8

Gozel, 2020-11-16:
Hi Steve and Daniel, yes, that's it. Thanks! Apologies for having overlooked that. 
Question, will there be support for the new accelerometer?

daniel-thompson, 2020-11-16:
I don't think anybody is currently working on this, so don't expect anything in the short term. Ultimately we never know when an interested person will step forward to write something so, in the longer term, I just don't know.

Gozel, 2020-11-16:
Fair. 
Many thanks!