wasp-os/wasp-os #107: Is there any community / support chat?

Kozova1, 2020-11-26:
Using discord / matrix / telegram / IRC or whatever.

daniel-thompson, 2020-11-26:
@panos asked about this a few months back (in #62) and I reacted by setting up a [#wasp-os](https://webchat.freenode.net/#wasp-os) IRC channel at freenode.net.

I still remain a bit skeptical that the community is large enough for the channel to be effective but I notice there are still two or three people hanging out there most of the time so maybe there's be enough for a chat. I've added it to the README and we'll see what happens.

closed 2020-11-26 by daniel-thompson

panos, 2020-11-27:
@daniel-thompson A bridge could help!

Izaic, 2020-12-01:
There's also the pine64 pinetime discord channel which is bridged accross matrix, irc, and telegram.