wasp-os/wasp-os #110: Cannot start bluetooth advertisement

Gozel, 2020-11-26:
Hi Daniel, 
I am trying to advertise heart rate data via bluetooth using ubluepy from micropython. I followed this example [ubluepy_temp.py](https://github.com/daniel-thompson/micropython/blob/d198d2907a0837cb82f4d38984bb2ba068cacf77/ports/nrf/examples/ubluepy_temp.py). 

However, the app crashes at the advertise() function with either of the following errors: "OSError: cannot apply advertisement data status 0x08" or "OSError: cannot start advertisement. status 0x12"  

I don't know how to resolve this. Any help would be much appreciated. 

```python
import wasp
import icons

from ubluepy import Service, Characteristic, UUID, Peripheral, constants

uuid_health_sense = UUID("0x180D") # Health Device
uuid_hr = UUID("0x2A37") # Heart Rate characteristic

class MyBluetoothApp(): 
    NAME = 'MyBluetooth'

    def __init__(self):
        """Initialize the application."""
        pass
        
    def foreground(self):
        """Activate the application."""
        self._draw()

        self._notify_enabled = False
        self._serv_env_sense = Service(uuid_health_sense)

        hr_props = Characteristic.PROP_NOTIFY | Characteristic.PROP_READ
        hr_attrs = Characteristic.ATTR_CCCD
        self._char_hr = Characteristic(uuid_hr, props = hr_props, attrs = hr_attrs)

        self._serv_env_sense.addCharacteristic(self._char_hr)

        self._periph = Peripheral()
        self._periph.addService(self._serv_env_sense)
        self._periph.setConnectionHandler(self._event_handler)

        ## OSError: cannot apply advertisement data status 0x08 
        ## OSError: cannot start advertisement. status 0x12
        self._periph.advertise(device_name="P8", services=[self._serv_env_sense])
```

daniel-thompson, 2020-11-30:
I've not really tried using ubluepy directly. All the wasp-os tools (both wasptool and the gadgetbridge port) work by connecting to the REPL over Nordic UART Service and this is how I collect data from the device.

daniel-thompson, 2020-11-30:
@Carglglz has done a few experiments in this area although I think there were some reliability issues with that code too.

Gozel, 2020-12-01:
Thanks, I will check it out.

closed 2020-12-08 by Gozel

Carglglz, 2021-01-12:
>      ## OSError: cannot apply advertisement data status 0x08 
>      ## OSError: cannot start advertisement. status 0x12
>       self._periph.advertise(device_name="P8", services=[self._serv_env_sense])

This is because with NUS service enabled, the device is already advertising at that point, so you need to stop it first and then advertise again.

> I am trying to advertise heart rate data via bluetooth using ubluepy from micropython

My recommendation is to use esp32 (MicroPython) which at this point its bluetooth low energy stack is pretty stable and easy to develop with. 
In the future when nrf port gets more stable it should be pretty easy to port all the work you may have done by then. 👍

jones139, 2021-05-08:
> This is because with NUS service enabled, the device is already advertising at that point, so you need to stop it first and then advertise again

I am just trying to do this, but am not sure how.   Calling (from python) ubluepy.Peripheral().advertise_stop() still gives me the same error - so I suspect that the NUS service is running on a different instance of Peripheral.

I have found that in micrpython/ports/nrf/main.c there is a call to ble_uart_init0(), which is defined in micropython/ports/nrf/drivers/bluetooth/ble_uart.c.    ble_uart_init0() starts the advertising (from C).    

Is the best way to inhibit this starting in main.c and try to start the ble uart from python along with my services - or is there another way?  (sorry, I am not used to micropython!).

Thanks

Graham.

rennard, 2021-08-02:
> This is because with NUS service enabled, the device is already advertising at that point, so you need to stop it first and then advertise again.
> My recommendation is to use esp32 (MicroPython) which at this point its bluetooth low energy stack is pretty stable and easy to develop with.
> In the future when nrf port gets more stable it should be pretty easy to port all the work you may have done by then. 👍

Could you expand on this at all please? Running into the same issue as jones139 that it doesn't seem possible to stop the advertising via micropython without a reference to the peripheral instance, though I'm happy to be proven wrong. If this is the case, any signposting on how else to approach this problem would be much appreciated. Thanks

Carglglz, 2021-08-03:
@rennard The device starts advertising in this line in `watch.py` 
https://github.com/daniel-thompson/wasp-os/blob/15434a38411ce49e3250e966ffd2472efe65f5df/wasp/boards/p8/watch.py.in#L36
See my PR #81, where I implemented a Time Service in [`wasp/bleprofile.py`](https://github.com/daniel-thompson/wasp-os/pull/81/files#diff-9df36eb81815523135e311b966b3c9aa24e98876d66e7457a23e829e4b68d837)

So you could stop advertising first or comment out the line in `watch.py`, either way should work. 👍

rennard, 2021-08-05:
@Carglglz thanks a lot for this, definitely helps me get on the right path. I've only just started developing with wasp-os so forgive me if this is a stupid problem (pretty sure it is) but I've accidentally disabled the UART advertising.

I want to undo this and upload a file that undoes my updates, so I can advertise both UART and a second service, or preferably just wipe any files on the flash and go back to a clean slate. I've tried going into DFU mode and uploading micropython.zip, this successfully uploads but my changes are still there when it reboots. Every time I try and run wasptool, upload/console/rtc the device vibrates and reboots, any ideas? How did you upload your file to the device without nuking the UART service? Thanks a lot for your help

Carglglz, 2021-08-05:
@rennard 
> How did you upload your file to the device without nuking the UART service?

I didn't, any modifications I did I compiled into micropython.zip and flashed through DFU. 

> so I can advertise both UART and a second service

This is not possible right now, there is not enough memory in bluetooth stack to add a second service 🤷‍♂️ . (To do this you should modify the ble stack in micropython nrf port).

>  Every time I try and run wasptool, upload/console/rtc the device vibrates and reboots, any ideas?

Did you upload a custom `main.py` or `watch.py`? 🤔 In that case I think micropython runs files in flash over those with the same name frozen in the firmware. (micropython.zip)

rennard, 2021-08-15:
@Carglglz Thanks for this, you pointed me in the right direction and I'm much closer to having what I want now, much appreciated! 

> This is not possible right now, there is not enough memory in bluetooth stack to add a second service 🤷‍♂️ . (To do this you should modify the ble stack in micropython nrf port).
I'm not sure I'll do this, but out of interest, could you give me a point in the right direction on how to do this?