wasp-os/wasp-os #113: Cannot access rtc.set_localtime in apps

chenguokai, 2020-12-02:
I was trying to write an App that adjust global time on device. But if I want to call `set_localtime`, the interpreter complains
```
AttributeError: 'RTC' object has no attribute 'set_localtime'
```

the fauly def is attached below:

```python
    def press(self, button, state):
        """Notify the application of a button-press event."""
        draw = wasp.watch.drawable
        if self.stage == 4:
            draw.string('Time set done', 0, 108, width=240)
            wasp.watch.rtc.set_localtime((self.year, self.month, self.day, self.hour, self.minute, 0))
            self.stage = 0
        else:
            self.stage = self.stage + 1
```

tht-jxny, 2020-12-11:
I had the same problem (See #73). The function does not appear in RTC.__dict__. It would be great if this problem is solved

Kozova1, 2020-12-14:
I do not know how to solve this, but my fork has a `set-rtc-113` branch that has a self test for that, might help find the cause.

https://github.com/Kozova1/wasp-os/tree/set-rtc-113

daniel-thompson, 2020-12-24:
watch.rtc.set_localtime() is working for me (I tested it with GadgetBridge and via `./wasptool --console`).

tht-jxny, 2020-12-24:
Is it also working if you add this piece of code into one of the application?

tht-jxny, 2020-12-24:
Concerning GadgetBridge: Is there any documentation on how to set up GadgetBridge so it works with the PineTime? @daniel-thompson

daniel-thompson, 2020-12-24:
> Concerning GadgetBridge: Is there any documentation on how to set up GadgetBridge so it works with the PineTime? @daniel-thompson

Not at the moment. However it as simple as _just click (+) and choose the wasp-os device from the resulting list_.

Make sure you use the wasp-os fork at https//github.com/daniel-thompson/gadgetbridge since the code is not yet in the upstream version.

tht-jxny, 2020-12-24:
Thanks for this advice! However, could you please include an .apk file in the gadgetbridge repo since not everybody may have the chance to build it for their own with Android Studio?

daniel-thompson, 2020-12-24:
> Thanks for this advice! However, could you please include an .apk file in the gadgetbridge repo since not everybody may have the chance to build it for their own with Android Studio

Check the Actions tab...

tht-jxny, 2020-12-24:
Thanks a lot and also for all your effort! Merry Christmas ;)

daniel-thompson, 2020-12-26:
I can't get the code from @Kozova1 to run.

It looks like it hasn't been tested on a real device (and since set_localtime() is not implemented on the simulator time setting code does need to be tested on a real device).

tht-jxny, 2020-12-26:
@daniel-thompson You don't know how much this piece of information means to me! I implemented the `set_localtime` function into my calculator app, ignored any error message in the simulator and flashed it to my PineTime and it works! :D

Even if this should not be done and despite the fact that it is dirty, I have to say that I love to set the rtc with my calculator! Thanks a lot!!
This issue should now be solved...

Kozova1, 2020-12-27:
Shouldn't the simulator at least provide stubs for `set_localtime`? Just print something like: `Set time to: $TIME`

closed 2021-01-30 by daniel-thompson