wasp-os/wasp-os #118: Add Line drawing function to draw565 library

Kozova1, 2020-12-06:
I noticed that the drawing library apparently does not support lines.
Is it possible to add a line drawing function?

I tried hacking in a line function but couldn't figure out how to write directly to the screen.

Kozova1, 2020-12-06:
Ideally, it would have an API like this:

`watch.drawable.line(x1,y1,x2,y2, color)`

maybe even add a thickness option?

daniel-thompson, 2020-12-06:
No objections here. One of the Bressenham variations should work fine.

Kozova1, 2020-12-10:
If you can help me understand how raw writing to the display works (quick_write etc.) I'll get a PR for it.

daniel-thompson, 2020-12-10:
quick_write() (which is designed to stop you having to keep pointlessly toggling the chip select) won't do you any good here because you can only write a couple of bytes at a time. You'll need to use display.set_window() to select the "rectangle" you want to update (start of with a 1x1 rectangle) and then use display.write() to draw the pixel.

Note drawing angled lines directly into the video RAM will be very slow because there's pretty much no way to optimise the transfer of data.
It will be very slow...

Kozova1, 2020-12-11:
see pr #124

closed 2020-12-12 by daniel-thompson