wasp-os/wasp-os #119: Add support for other languages

Kozova1, 2020-12-06:
Currently, characters in other languages (Hebrew for example) appear as question marks.

Ideally, you'd be able to choose which languages you want support for (this should only be a font change),
and build wasp-os respectively.

Kozova1, 2020-12-07:
see this PR in the InfiniTime repo https://github.com/JF002/Pinetime/pull/128#issuecomment-723573463

daniel-thompson, 2020-12-10:
I think the main missing pieces (at least for languages with relatively small alphabets) this are some means to substitute translated versions of the strings and (probably) some means to encode discontinuous character codes without wasting space (e.g. improving or relplacing tools/micropython-font-to-py).

For right-to-left languages like Hebrew there will be some additional plumbing needed because currently the drawing API doesn't require apps to tell it where the right margin is.

... after that it is "just" the build system.

Kozova1, 2020-12-11:
> Efficient support is now provided for sparse character sets.
The second problem seems to be solved for tools/micropython-font-to-py

daniel-thompson, 2020-12-11:
> The second problem seems to be solved for tools/micropython-font-to-py]

Great. That would make the task a little easier.

The font decoder in wasp-os is customized (to make use of viper codegen) and IIRC this is hard coded for continuous character codes... but it follows the same structure as the autogen code from micropython-font-to-py so adopting the search logic shouldn't be too hard.