wasp-os/wasp-os #121: App Idea: Calculator Application

Izaic, 2020-12-08:
As far as i can tell there is no calculator app for wasp-os yet. I think I'll be up to doing it sometime for fun, however i am putting the idea here if someone else wants a neat and useful project to work on too.

The idea would be for it to be a very very basic calculator with a numpad of 0-9 with multiplication, division, subtraction, and addition buttons, a clear button, and perhaps a decimal button. At the top of the screen in plain numbers it would display the answers. Nothing special, but definitely could be useful in a pinch for double checking large numbers. Keeping it as simple as possible would be best for space and ram usage reasons.

SteveAmor, 2020-12-08:
Hi @Izaic @tht-jxny created a calculator app that can be found here https://github.com/tht-jxny/MicroPineTime

Izaic, 2020-12-08:
It would be awesome to see this merged into waspos as an official application.

tht-jxny, 2020-12-11:
Of course, I can open a new pull request! But currently, every button is an image and I consider the colors of the buttons to be somewhat awkward. (Have a look at the picture in #22) Do you have any possible improvements in your mind? I did also temporarily include a function to set the rtc when a specific pattern is entered.

Izaic, 2020-12-12:
> Of course, I can open a new pull request! But currently, every button is an image and I consider the colors of the buttons to be somewhat awkward. (Have a look at the picture in #22) Do you have any [...]

Oh wow! Very cool idea with the RTC clock! I think that should stay in since it means you wont have to sync the watch with a computer running wasptool to set the time.

That said, i think keeping it as simple as possible is good... I don't think it's too important to have an advanced calculator on your wrist seeing as you can always grab your phone... However having a basic calculator for decimals, multiplication, addition, subtraction and division would be very handy in a pinch.

Not that I'm dissing the parenthesis and exponent buttons, but with the memory constraints of the Pinetime, every little non-essential thing should be kept in mind especially with how little space sounds like is left. Perhaps just render everything in text, drop the parenthesis, exponent carot, and equals button. When you enter in a number and press a operator it clears out and awaits the next number, and then shows the answer unless you press another operation key.

[image]

daniel-thompson, 2020-12-12:
Assuming the calculator works by collecting the key presses and passing them to eval() then stripping out features like brackets will not make any difference to RAM usage when the app is in the background (which is the most important thing to reduce) so I don't see much point in removing these.

I do agree it would be better to change the "one big image" to a button graphic (whose colour you can shift with palette effects) and then render text over the top.

Finally using the calculator to enter the time is a neat hack but, as @tht-jxny hinted it's not "the right thing to do" and I wouldn't want that in the main source base.

Izaic, 2020-12-12:
Fair enough. I still think the parenthesis and whatnot might want to be stripped out simply for physical screen real estate... It looks mighty crowded in that picture in #22... Unless there's ideas to improve that?

daniel-thompson, 2020-12-12:
Yes, if there were to be a useability boost from dropping features than that might be different. To be fair it is also an excellent idea to think carefully about the tradeoff between features and RAM usage (especially background RAM usage). It's simply that in this instance the features didn't really impact that.

I also note that it's pretty rare to have calculators with a 4x4 button grid. Calculators usually need an extra row to implement *either* memory functions (like the old 20 button Casio calculators) or brackets otherwise there are lots of expressions that you can't compute. I won't veto either approach though if it is accompanied by a good useability story.

tht-jxny, 2020-12-12:
I've been following the discussion for a while now and thought I'd just post my code so we can use this as a base for further discussion and improvement.
As for how to enter the time, I would say that this is indeed not part of the main source base.  Nevertheless, I'd like to know why it didn't work. ( #113 describes the problem I also had)

A few remarks about my code:
- function names should look like `my_function()`
- There is the question if I should store an image for each and every button or draw all buttons as one image
- Draw buttons as text and not as images?

My code:
``` python
# 2-bit RLE, generated from res/calc.png, 413 bytes
calc = (
    b'\x02'
    b'`@'
    b'(@\x03P?\x0eV?\x08[?\x04_?\x01b'
    b'<f9h7j5l3n1p/r-t'
    b'+H\xd1D\xd1H*H\xd2B\xd2H)I\xd2B'
    b"\xd2I'J\xd2B\xd2I'J\xc8\xc2\xc8B\xd2J"
    b'%K\xc7\xc1\x02\xc1\xc7B\xd2J%K\xc7\xc1\x02\xc2'
    b'\xc6B\xd2K$K\xc5\xc2\xc1\x02\xc1\xc2\xc5B\xd2K'
    b'#L\xc4\xc1\x08\xc1\xc4B\xc5\x08\xc5L"L\xc4\xc1'
    b'\x08\xc2\xc3B\xc5\x08\xc1\xc1\xc3L"L\xc5\xc2\xc1\x02'
    b'\xc8B\xcf\xc1\xc2L!M\xc6\xc2\x02\xc8B\xd0\xc1\xc1'
    b'M M\xc7\xc1\x02\xc8B\xd1\xc1M M\xc8\xc2\xc8'
    b'B\xd2M M\xc9\xc1\xc8B\xd2M M\xd2B\xd2'
    b'M M\xd2B\xd2M N\xd0D\xd0N \x7f\x01'
    b' \x7f\x01 N\xd0D\xc1\xcfN M\xd2B\xc1\xd1'
    b'M M\xd2B\xd2AL M\xd2B\xd2AL '
    b'M\xd2B\xd2AL M\xc7\xc1\xcaB\xc5\xc8\xc5A'
    b'L M\xc5\xc1\x02\xc2\x02\xc1\xc5B\xc4\xc1\x08\xc1\xc4'
    b'AL!L\xc5\xc1\x06\xc1\xc5B\xc4\xc1\x08\xc1\xc1\xc3'
    b'AK"L\xc6\xc1\x04\xc1\xc6B\xc6\xca\xc2AK"'
    b'L\xc6\xc1\x04\xc1\xc6B\xc7\xca\xc1L#K\xc5\xc1\x06'
    b'\xc1\xc5B\xc4\xc1\x08\xc1\xc4JA$K\xc5\xc1\x02\xc2'
    b'\x02\xc1\xc5B\xc4\xc1\x08\xc1\xc4K%J\xc6\xc2\xc2\xc2'
    b"\xc6B\xc5\xc8\xc5J&J\xd2B\xc6\xccJ'I\xd2"
    b'B\xc7\xcbI(I\xd2B\xc8\xcaI)H\xd2B\xc9'
    b'\xc9H*H\xd1D\xc9\xc8H+t-r/p1'
    b'n3l5j7h9f<b?\x01^?\x05'
    b'XAA?\tV?\x0eP('
)
# 2-bit RLE, generated from res/calc_buttons/calc_num_0.png, 677 bytes
calc_num_0 = (
    b'\x02'
    b'44'
    b'?8@\xf4U\x1dY\x19]\x16`\x12c\x10e\x0e'
    b'N\x80\xd0\x81\xc0\x88\xc1@DA\x80 \x81\x01\x01\x01'
    b'\x81A\xc0\x8c\xc1@\xf0A\x80\xf4\x8e\r\x8c\xc0\xd0\xc1'
    b'@DA\x01\t\x01\x80d\x81\xc0\xf0\xc1@\xf4M\x0b'
    b'KA\x80\x88\x81\x01\r\x01\xc0\x8c\xc1L\nKA@'
    b'DA\x11\x80h\x81\xc0\xf4\xcc\t\xca\xc1A\x13\x81\xcc'
    b'\x07\xcb\x81\x06\x01A@\xacA\x80\xf0\x81\x81\xc0\xd0\xc1'
    b'A@$A\x07\x80\x8c\x81\xc0\xf4\xcb\x07\xca@\xacA'
    b'\x06\x80 \x81\xc0\xd0\xc1@\xf4G\x80\x88\x81\x01\x05\x01'
    b'\xc1K\x05JA\x01\x05\xc0 \xc1@\xf0A\x80\xf4\x89'
    b'\xc0\xac\xc1\x01\x05@DA\x8b\x05\x8a\x80\x8c\x81\x05\x01'
    b'\xc0\xd0\xc1@\xf4K\x80\x88\x81\x06\xc1K\x04J\xc0 '
    b'\xc1\x05@hA\x80\xf4\x8c\x81\xc1\x05\xc0d\xc1\x8b\x04'
    b'\x89@\xccA\x05\x01\x80\xf0\x81\xc0\xf4\xcd@\xacA\x05'
    b'\x01\x81\xca\x04\xc9\x80h\x81\x05\xc0D\xc1@\xf4N\x80'
    b'\xf0\x81\x01\x05\xc0\xac\xc1J\x04I@$A\x05\x80\x8c'
    b'\x81\xc0\xf4\xcf@DA\x05\x80h\x81\xca\x04\xc8\xc1\x01'
    b'\x05\xc0\xd0\xc1@\xf4O\x80\x88\x81\x05\xc0D\xc1J\x04'
    b'H@\xf0A\x05\x01\x80\xf4\x81\x8f\xc0\xac\xc1\x05@ '
    b'A\x8a\x04\x88\x80\xd0\x81\x05\x01\xc0\xf4\xd0\x81\x05\x01\xca'
    b'\x04\xc8@\xccA\x05\x80 \x81\xd0\xc0\xf0\xc1\x05\x01@'
    b'\xf4J\x04H\x80\xac\x81\x05\xc0 \xc1P@\xf0A\x05'
    b'\x01\x80\xf4\x8a\x04\x88\xc0\xd0\xc1\x05\x01\x90\xc1\x05\x01\x8a'
    b'\x04\x88A\x05\x01\x81\x8f@\xacA\x05\x80 \x81\xc0\xf4'
    b'\xca\x04\xc8\xc1\x01\x05@\xd0A\xcf\x80\x88\x81\x05\xc0D'
[... 691 lines cut ...]
                    time_list = [int(x) for x in self.output.split(".")]
                    # Debugging: print(tuple(time_list))
                    wasp.watch.rtc.set_localtime(tuple(time_list))
            if len(self.output) >= 11:
                wasp.watch.drawable.string(self.output[len(self.output)-11:], 5, 8, width=190)
            else:
                wasp.watch.drawable.string(self.output, 5, 8, width=190)

    def calculate(self,s):
        equation = s
        
        # Normal calculator stuff    
        for i in range(len(s)):
            if (s[i] =="^"):
                equation = s[:i] + "**"+s[i+1:]
            elif (s[i] == ":"):
                equation = s[:i] + "/"+s[i+1:]
        
        try:
            result = eval(equation)
        except:
            result = "" # Error
        
        return str(result)
```

Izaic, 2020-12-12:
I tried installing this to my Pinetime for testing but found i didn't have enough free memory.

I'm guessing with the haptic alarm app enabled there really isn't enough ram.

With that said, perhaps that's something replacing the button images with just text could help with?

Also perhaps i could disable the internal wasp self test tool to help?

tht-jxny, 2020-12-29:
#137 closes this.

closed 2020-12-29 by daniel-thompson