wasp-os/wasp-os #123: Wasp OS user interface

Izaic, 2020-12-08:
I am really enjoying how much movement Wasp OS has had lately! With that said, as the amount of apps for Wasp OS increases, I find it a bit more annoying to swipe sideways through each app... it would be cool to see some kind of app pinning system in a future Wasp-OS release, where you can pin important apps you use to the main swipe carousel, and then have the drop down menu (which shows the light feature & debug tool) list the other apps you may have installed so that you can tap on them and launch them. Perhaps this could also tie into the idea of dynamically loading apps? (Issue [122](https://github.com/daniel-thompson/wasp-os/issues/122#issuecomment-740390393))

Perhaps this is something for the roadmap?

daniel-thompson, 2020-12-10:
I don't really understand what you mean. By default there are only four apps in the quick ring (clock, heart rate, stopwatch, step counter). Everything else is started using the launcher.

Izaic, 2020-12-10:
Hm, i didn't know, sorry. I haven't tested any of the apps yet because of the python issue. Going to close this.

With that said, are apps loaded into ram only when launched?

closed 2020-12-10 by Izaic