wasp-os/wasp-os #125: Step Counter not working

SecureCodeJIT, 2020-12-13:
updated sealed unit everything else works.  steps counter does not update.

daniel-thompson, 2020-12-13:
Which watch model?

SecureCodeJIT, 2020-12-14:
Last night steps started working don't understand the delay.  not sure model bought a couple of months ago.

daniel-thompson, 2020-12-14:
Surely you know what type of watch you actually bought though?
https://wasp-os.readthedocs.io/en/latest/install.html#device-support

closed 2020-12-23 by daniel-thompson

pkubaj, 2021-02-08:
I have a PineTIme (sealed) and I think I can confirm this issue.

Right after flashing WaspOS (tag 0.3), I walked into another room and the step counter showed some steps, but then stopped updating when I walked around the house.

The next day, however, it worked fine and seems to work fine now (it's the day after now).