wasp-os/wasp-os #126: Notifications

SecureCodeJIT, 2020-12-13:
Improvement

- link to cell phone for text and email notifications

Izaic, 2020-12-13:
I'd imagine this is on the gadgetbridge side of things, not WaspOS itself. That said i agree!

Kozova1, 2020-12-14:
A custom gadgetbridge build exists that supports it - https://github.com/daniel-thompson/gadgetbridge

SecureCodeJIT, 2020-12-14:
i use nRFconnect gadgetbridge is not working for me yet.

Kozova1, 2020-12-14:
Official gadgetbridge doesn't work - you have to build the one I linked above from source

daniel-thompson, 2020-12-24:
Agree. The custom Gadgetbridge has been available to use since July!

I have recently set up an autobuilder so you don't have to compile it yourself. Generally speaking you can go to the most recent build and then grab the arteacts from the bottom of the page. See:
https://github.com/daniel-thompson/gadgetbridge/actions?query=event%3Apush+

closed 2020-12-24 by daniel-thompson

pathcl, 2021-01-21:
@daniel-thompson what if you’re on iOS? or you want to send data using your own linux box? is it possible?. Use case: send notifications for gmail, prometheus, twitter, so on.