wasp-os/wasp-os #128: Update GadgetBridge fork to allow automatic RTC set

cr0mbly, 2020-12-14:
### Summary

This is an idea up for consideration in regards to limitations of GadgetBridge for our use cases of wasp-os. Annoyingly on connect between the app and device the app emits a 

```JS
\u0010setTime(<timestamp>);E.setTimeZone(<tz>);&&require('Storage').write('setting.json',s);})(require('Storage').readJSON('setting.json',1))
```

call which is missed by wasp-os's GadgetBridge watcherer because of the leading unicode `\u0010` and the fact that there's a bunch of other junk at the end that we don't need (the storage portion of the call). If we changed the call signature in 

`https://codeberg.org/Freeyourgadget/Gadgetbridge/src/branch/master/app/src/main/java/nodomain/freeyourgadget/gadgetbridge/service/devices/banglejs/BangleJSDeviceSupport.java::transmitTime` to something 

```
GB({'setTime': <timestamp> })
```

We could capture this event and allow the watch to update the time automatically on connect, the downside is that users of wasp-os would need to know to use the form of GadgetBridge that @daniel-thompson has already committed too but it would allow us to get this going, possibly in the future we might get actual support from the project and be able to define our own schema but this could be a good first stem, thoughts?

daniel-thompson, 2020-12-15:
I think this is already in the roadmap: https://wasp-os.readthedocs.io/en/latest/TODO.html

Note however that, in my view, hacking BangleJS adapter to make it incompatible with a real BangleJS devices is a non-starter. There's no way the upstream project would accept wasp-os changes if we do that. The plan is to introduce a fully fledged adapter for wasp-os devices.

cr0mbly, 2020-12-15:
> Note however that, in my view, hacking BangleJS adapter to make it incompatible with a real BangleJS devices is a non-starter. There's no way the upstream project would accept wasp-os changes if we do that. The plan is to introduce a fully fledged adapter for wasp-os devices.

Great thanks. Just wanted to get input if it was worth looking into. I guess one possibility instead of hacking into the bangleJS implementation we generate a `waspOS` version derived from it and see if Gadgetbridge would be happy with us pushing it [upstream](https://codeberg.org/Freeyourgadget/Gadgetbridge/src/branch/master/app/src/main/java/nodomain/freeyourgadget/gadgetbridge/devices), this would isolated from bangleJS but the original source would be based on it. Although that would require more maintenance as updates are made so it's probably not a great idea either.

cr0mbly, 2020-12-20:
Closing off as discussed above, a dedicated adaptor seems to be the way to go :+1:

closed 2020-12-20 by cr0mbly