wasp-os/wasp-os #129: Bluetooth API

shulltronics, 2020-12-15:
Hi everyone,

Thanks for the great work on wasp-os. I'm looking for some direction on how to use wasp-os to interact with the PineTime over Bluetooth. For example, if I want to implement a simple test application that prints characters to the PineTime screen from a Bluetooth connection with my laptop, how could I accomplish this? 

I'm getting the hang of writing my own applications, but it's not clear to me how to work with Bluetooth. 

Thanks for any direction you can give me.

cr0mbly, 2020-12-20:
Hi @shulltronics wasp-os communicates over bluetooth connections via [gadgetbridge](https://gadgetbridge.org/) application for data transmission, this dumps data via the bluetooths connection to the python REPL for notifications which is picked up with the `wasp-os/wasp/gadgetbridge.py::GB` class. this happens automatically via the underlying libs which is why you don't see any direct bluetooth code in wasp os. Take a look at @Carglglz PR which has a go at playing around with the underlying bluetooth library for RTC https://github.com/daniel-thompson/wasp-os/pull/81 it seems like the libs are a bit hit or miss at the moment, but should be a good starting point if your wanting to have a play around :)

shulltronics, 2020-12-28:
Hey @cr0mbly, thanks for getting back with me! I did some exploration before you replied and was able to get the BLE services from #81 working on my watch, so I'm in a good place to keep learning about it. Thanks for your help though! I'll close this issue.

closed 2020-12-28 by shulltronics