wasp-os/wasp-os #130: Help for use wasp simulator with Twatch 2020

picpic020960, 2020-12-19:
Hello Sactre and Daniel
sorry for my bad english (French)
wasp firmware for Twatch 2020 ok on my watch.
How to use the simulator for some update in python firmware ( Timer , Alarm)
I tested a few things but now I'm lost
Thanks for help

_Originally posted by @picpic020960 in https://github.com/daniel-thompson/wasp-os/issues/58#issuecomment-748448117_

daniel-thompson, 2020-12-23:
I'm afraid I don't understand. I'm get how the wasp simulator could be used on a Twatch 2020.

picpic020960, 2020-12-25:
Thanks for reply
this repo is for example : https://github.com/sactre/wasp-os
Twatch 2020 + wasp-os (fork) 
maybe someone else will have some ideas
Regards

daniel-thompson, 2020-12-26:
> maybe someone else will have some ideas

Maybe.

What the actually question is remains very unclear!

The wasp-os simulator is one type of (simulated) device and and the Twatch 2020 is a completely different device. You cannot do Twatch 2020 development using the simulator. What is the link between them?

picpic020960, 2020-12-28:
Twatch 2020 is ESP32 with micropython. 
see this example https://github.com/sactre/wasp-os : Twatch 2020/ESP32 with wasp-os
but without simulator. it's why my question.
Twatch  hardware is better than Pinetime...

daniel-thompson, 2021-01-30:
To be honest I'm still clear what the questons is since there is no link between the simulator and the Twatch 2020 ports. Howeverit might be better to raise this on sactre's fork anyway so I'm going to close this at this point.

closed 2021-01-30 by daniel-thompson