wasp-os/wasp-os #138: Alarm App Fault

tht-jxny, 2020-12-29:
I was modifying the AlarmApp to my needs so that an alarm repeats over and over again after some defined interval (Useful for workouts). 
By doing so, I noticed that the alarm won't go off when the app is still in the foreground. Should this be corrected (Probably rather in the system logic) or is this fine for normal use?

Shall I open a pull request for my "Interval Timer" at all? Or shall I, due to the similarities with the normal alarm app, just modify the existing application and include a switch that toggles between these two functionalities?

daniel-thompson, 2020-12-29:
What you are describing is part of the design of the alarm app. The alarm is disarmed when the foreground() method is called and only activated when the background()  method is called. This avoids the alarm acidentally triggering if you touch the spinners and take it through the current time.

I'm sort of interested in a pull request for an interval timer. There is a discussion in another thread (#134 with @BNolet) about a activity/sports app with where either a stopwatch or countdown timer is shared with a "trip counter" on the pedometer. In my view an interval timer it likely to end up as part of that. However it's not problem to have it start out as a standalone application... just be aware that the app will liikely be subject to very heavy modding if/after it is merged.

BNolet, 2020-12-29:
I'd be partial to including an interval timer in there

daniel-thompson, 2021-01-16:
I think we've cover the quesitons in the original comment so I'll close at this point.

closed 2021-01-16 by daniel-thompson