wasp-os/wasp-os #140: wasp os installation for dummies?

masterdelic, 2020-12-30:
hello,

i'm trying to flash wasp os on a pine time since two days with no luck. i am not a programmer and i didn't understand many steps. where are the reloader-mcuboot.zip and the micropython.zip? how to generate them from building wasp os from source?  can you please make an installation guide for dummies? or just share those files to make it work?

thanks in advance

Kozova1, 2020-12-30:
You can download a build from the releases tab, the `wasp-os-0.3.tar.gz` file contains everything you need.

masterdelic, 2020-12-31:
thanks a lot, i finally make it work!

closed 2021-01-01 by masterdelic

masterdelic, 2021-01-05:
hello, 
dummies are back... 
How am i going to enable all the apps like daniels last video?
if i bother you it's ok i understund, just tell me to sell my pine time...

Kozova1, 2021-01-05:
The build you downloaded is (probably) an old version. To get all of the apps that are shown in that video you'd need to download a build from the CI.

daniel-thompson, 2021-01-05:
> if i bother you it's ok i understund, just tell me to sell my pine time...

I hope noone ever says that to you. That would be a betrayal of both the PineTime and wasp-os communities.

There is a link to the latest binaries in the video's description.

masterdelic, 2021-01-05:
i just did it!!!
it's awesome! 
great job!
many congrats!!!

masterdelic, 2021-01-05:
thanks a ton!

Izaic, 2021-01-05:
> hello,
> dummies are back...
> How am i going to enable all the apps like daniels last video?
> if i bother you it's ok i understund, just tell me to sell my pine time...

While i personally would never recommend someone to buy a Pine product if they don't have an inkling of an idea of what it will do if they aren't willing for the money to be lost if they don't end up liking it... Nobody should be saying to get rid of one, as you can always learn.

masterdelic, 2021-01-05:
Even if all of you tell me to sell it I wouldn't. Those quarantine days I'm reading too much, I have a strong will and slowly slowly  I'll find all my "bugs".  Thanks again!!!

jollyblondgiant, 2022-01-09:
Hi, I don't know exactly what process masterdelic used to overcome this obstacle, but I am certainly feeling like a dummy trying to flash wasp-os onto my pinetime. 

`$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.3 LTS
Release:	20.04
Codename:	focal

$ cd ~/wasp-os
$ ls build-pinetime/
bootloader-daflasher.zip  bootloader.hex  micropython.zip  reloader-factory.zip  reloader-mcuboot.zip  reloader.zip
$ sudo hcitool lescan
LE Scan ...
C3:7F:47:5F:08:1F InfiniTime
[... & others]
[in another tab, I connect and pair to pinetime with bluetoothctl]:
# info  C3:7F:47:5F:08:1F
Device C3:7F:47:5F:08:1F (random)
	Name: InfiniTime
	Alias: InfiniTime
	Appearance: 0x00c2
	Paired: yes
	Trusted: yes
	Blocked: no
	Connected: yes
	LegacyPairing: no
	UUID: Vendor specific           (00000000-78fc-48fe-8e23-433b3a1942d0)
	UUID: Vendor specific           (00001530-1212-efde-1523-785feabcd123)
	UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
	UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
	UUID: Immediate Alert           (00001802-0000-1000-8000-00805f9b34fb)
	UUID: Current Time Service      (00001805-0000-1000-8000-00805f9b34fb)
	UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
	UUID: Heart Rate                (0000180d-0000-1000-8000-00805f9b34fb)
	UUID: Battery Service           (0000180f-0000-1000-8000-00805f9b34fb)
	UUID: Alert Notification Serv.. (00001811-0000-1000-8000-00805f9b34fb)
	UUID: Vendor specific           (00010000-78fc-48fe-8e23-433b3a1942d0)
 tools/ota-dfu/dfu.py -z build-pinetime/micropython.zip -a  C3:7F:47:5F:08:1F --legacy
Sending file firmware.bin to C3:7F:47:5F:08:1F
Binary imge size: 335764
Binary CRC32: 1060317325
Connecting to C3:7F:47:5F:08:1F
Couldn't connect, will try DFU MAC
Connecting to C3:7F:47:5F:08:20
Exception at line 164: Can't connect to device
[then i restart pinetime by holding the button for ~5s and releasing as soon as the pinecone graphic appears]
$ tools/ota-dfu/dfu.py -z build-pinetime/micropython.zip -a  C3:7F:47:5F:08:1F --legacy
Sending file firmware.bin to C3:7F:47:5F:08:1F
Binary imge size: 335764
Binary CRC32: 1060317325
Connecting to C3:7F:47:5F:08:1F
Need to switch to DFU mode
Exception at line 154: UUID not found: 00001531-1212-efde-1523-785feabcd123
[make sure I'm connected in the bluetoothctl tab]:
# info  C3:7F:47:5F:08:1F
Device C3:7F:47:5F:08:1F (random)
	Name: InfiniTime
	Alias: InfiniTime
	Appearance: 0x00c2
	Paired: no
	Trusted: yes
	Blocked: no
	Connected: no
	LegacyPairing: no
	UUID: Vendor specific           (00000000-78fc-48fe-8e23-433b3a1942d0)
	UUID: Vendor specific           (00001530-1212-efde-1523-785feabcd123)
	UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
	UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
	UUID: Immediate Alert           (00001802-0000-1000-8000-00805f9b34fb)
	UUID: Current Time Service      (00001805-0000-1000-8000-00805f9b34fb)
	UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
	UUID: Heart Rate                (0000180d-0000-1000-8000-00805f9b34fb)
	UUID: Battery Service           (0000180f-0000-1000-8000-00805f9b34fb)
	UUID: Alert Notification Serv.. (00001811-0000-1000-8000-00805f9b34fb)
	UUID: Vendor specific           (00010000-78fc-48fe-8e23-433b3a1942d0)
# pair  C3:7F:47:5F:08:1F
Attempting to pair with C3:7F:47:5F:08:1F
[CHG] Device C3:7F:47:5F:08:1F Connected: yes
Request authorization
[agent] Accept pairing (yes/no): [CHG] Device C3:7F:47:5F:08:1F ServicesResolved: yes
[agent] Accept pairing (yes/no): yes
[CHG] Device C3:7F:47:5F:08:1F Paired: yes
Pairing successful
# connect  C3:7F:47:5F:08:1F
Attempting to connect to C3:7F:47:5F:08:1F
Connection successful
$ tools/ota-dfu/dfu.py -z build-pinetime/micropython.zip -a  C3:7F:47:5F:08:1F --legacy
Sending file firmware.bin to C3:7F:47:5F:08:1F
Binary imge size: 335764
Binary CRC32: 1060317325
Connecting to C3:7F:47:5F:08:1F
Couldn't connect, will try DFU MAC
Connecting to C3:7F:47:5F:08:20
Exception at line 164: Can't connect to device

`
What steps am i missing? why can't I connect to my device with dfu.py when I can connect and even set time from command line?
thank you, pineOs wizards, for hearing my plea