wasp-os/wasp-os #142: Improve ST7789 display driver performance

miguelrochefort, 2021-01-03:
Rendering on wasp-os is slow because the ST7789 display driver is implemented in MicroPython. Replacing it with a C implementation would improve rendering performance.

This ST7789 driver implemented in C seems like a good place to start: https://github.com/russhughes/st7789_mpy

Here's the kind of performance improvement we can hope for (click to open video):

[image]

* Has this been attempted before?
* How much work will it take?
* What are possible challenges?

Kozova1, 2021-01-03:
I think that [this](https://docs.micropython.org/en/latest/develop/cmodules.html) might be one of the better ways to do this using C.

daniel-thompson, 2021-01-03:
Personally I'd expect optimizing draw565 as a C library or redesigning the micropython SPI interfaces to allow double buffering would likely yield much bigger wins than rewriting the st7789 driver. However that's partly a guess (although it is a guess based on the ineffectiveness of python level optimizations to the existing st7789 driver) so ultimiately it will come down to the benchmarks. There are lots of benchmarks built into the Self Test tool.

One trick would be to mark up all the existing st7789 methods with @micropython.native which will make them run roughly twice as fast (at the expense of code size). You could use those gains to estimate the maximum possible gains in a rewrite.

Kozova1, 2021-01-04:
The driver linked above by @miguelrochefort implements both the driver and some draw565 capabilities - notably line, fill, text etc.

Kozova1, 2021-01-05:
After some ~trial and error~ development with the library, I've gathered these things about it:
- It currently uses the same name (from the python side) `import st7789`.
- It seems to use `malloc` and `free` in some places, which aren't defined for the pinetime (or at least it seems so).
- The only error I get when including it into the micropython build used in wasp-os (currently) is a linker fail, since it can't find neither `malloc` nor `free`

daniel-thompson, 2021-02-20:
Drawing performance has been improved significantly since this ticket was raised. I think the optimizations from mid-Jan optimizations result add up to around 30% quicker for many benchmarks.

I'm going to close this at this point since discussion seems to have stopped and I am not aware of anyone working any further performance improvements.

To be clear there are certainly further optimizations possible whilst keeping the code python based (like fixing the micropython SPI API to permit double buffering) and it might also be possible to squeeze out a bit more performance by implementing parts of draw565 or ST7789 in C.

Note that no matter what tricks are attempted we will never see improvements like the one shown in the video though: the nRF52832 just can't clock the SPI bus fast enough to push out pixels as quick as the left hand side and I suspect that even though our driver is pure python we are already somewhat faster than the python code on the right (which is not to shabby given the CPU on the TTGO clocks 5x faster than nRF52832).

closed 2021-02-20 by daniel-thompson