wasp-os/wasp-os #144: GadgetBridge connection is not stable

miguelrochefort, 2021-01-06:
I have been using wasp-os on my Colmi P8 for over a week and I've noticed that the GadgetBridge connection is not stable:

- The watch disconnects from GadgetBridge within about an hour.
- The watch sometimes fails to receive notifications even if GadgetBridge indicates that it's connected.
- The music player freezes after sending too many successive commands (e.g., quickly toggle play/pause 10 times).

Is anyone else having these issues? Do we know what could be the cause?

mogorman, 2021-01-08:
i have same issue with music player.

miguelrochefort, 2021-01-10:
This is currently the biggest impediment in my adoption of wasp-os. I'm building 2 apps ([Home Assistant](https://www.home-assistant.io/), [Nomie](https://nomie.app/)) that rely on phone pairing, and they won't be usable as long as this problem remains.

If anyone has a hint about what's causing the issue and what to look for, I'd be willing to investigate.

daniel-thompson, 2021-01-10:
I suspect the problem to be in the low level code that implements the NUS protocol (which is part of micropython itself).

This commit is exploratory: https://github.com/daniel-thompson/micropython/commit/048bb2da83d0cbe73a829e461bd4f22b86c282bc

Before applying this patch when the NUS/BLE code failed the watch would reboot. With this change it hangs instead. Hardly an improvement but a definite sign that there are problems in this part of the code. I've no real idea why the TX code wedges... but it's definitely the code to be looking at.

I have sepculated about modifying the TX code to exit and retry later... but there's no evidence that this would actually unwedge anything.

PS It might also be useful to be working on hardware with the back off so you can hook up an SWD debugger to examine the state during a hang.

miguelrochefort, 2021-01-10:
> This commit is exploratory: daniel-thompson/micropython@048bb2d

Thanks. That would indeed explain the hangs.

However, it doesn't explain the other problem where GadgetBridge loses
connection with wasp-os (in which case there's no hang). Any idea what
might cause that?

Should we open 2 distinct issues, or do you think they might be related?

Sadly, I don't have a PineTime with the back off to use an SWD debugger, so
I won't be able to investigate this at the moment.

daniel-thompson, 2021-01-10:
> Should we open 2 distinct issues, or do you think they might be related?

Wouldn't suprise me if they are the same cause. Hard to say for certain until one of them is understood.

miguelrochefort, 2021-01-11:
On a related note, I have noticed that the watch produces 10-20 Bluetooth advertisements per second when unpaired and idle, using passive BLE scanning. In contrast, my Xiaomi Mi Band 3 only emits 1 advertisement per second.

I don't know if this could contribute to the above issues and/or negatively affect battery life, but I thought it was interesting to point out.

Carglglz, 2021-01-12:
> The music player freezes after sending too many successive commands (e.g., quickly toggle play/pause 10 times).

I warned about this in the PR #88 , and as @daniel-thompson says the problem is the NUS protocol / nrf port BLE stack. The reason why other notifications works vs music notifications crash is that Gadgetbridge protocol sends a considerable amount of data every time a play/pause/next/prev is issued by the watch. I guess this ends up causing a memory overflow in the RX buffer (NUS or BLE stack) and hence the reboot.

> On a related note, I have noticed that the watch produces 10-20 Bluetooth advertisements per second when unpaired and idle, using passive BLE scanning. In contrast, my Xiaomi Mi Band 3 only emits 1 [...]

I don't know how much but this does decrease battery life

I think you can set the advertisement interval in `micropython/ports/nrf/drivers/bluetooth/ble_drv.c` 
(Be aware I haven't test this yet. 🤷‍♂️ )
When it is already connected:
```
63 #define NON_CONNECTABLE_ADV_INTERVAL MSEC_TO_UNITS(100, UNIT_0_625_MS)
```
When it is not connected:
```
615     m_adv_params.interval                    = MSEC_TO_UNITS(100, UNIT_0_625_MS);
```
Although there is trade off, the less it advertises the harder is to find and connect to...

gebner, 2021-01-25:
> The watch sometimes fails to receive notifications even if GadgetBridge indicates that it's connected.

This seems to be related to the length of the notification.  (P8b here.)  You can reliably reproduce the issue by sending a notification with the text `long long long long long` from the debug tab in gadgetbridge:
```
$ adb logcat | grep UART
...: UART TX: GB({"t":"notify","id":1611516602,"subject":"long long long long long","body":"long long long long long","sender":"long long long long long","tel":"long long long long long"})
...: UART RX LINE: >>> GB({"t":"notify","id":16115162cg long long long long","body":"long long long long long","sender":"long long long long long","tel":"long long long long long"})
...: UART RX LINE: Traceback (most recent call last):
...: UART RX LINE:   File "<stdin>"
...: UART RX LINE: SyntaxError: invalid syntax for integer with base 10
```
It looks like long messages are garbled on transmission.

brvier, 2021-03-04:
> I don't know how much but this does decrease battery life

It s power consuming .... Strategy i use on nrf52 is to have fast adv after disconnect or power on and after a certain amount of time reduce to one by second.

brvier, 2021-03-04:
> This seems to be related to the length of the notification. (P8b here.) You can reliably reproduce the issue by sending a notification with the text `long long long long long` from the debug tab in [...]

> This seems to be related to the length of the notification. (P8b here.) You can reliably reproduce the issue by sending a notification with the text `long long long long long` from the debug tab in [...]

i've seen so many limitations on BLE stack on many phones, it s depends so much on BLE driver and constructor. For example some samsung device don 't let negociate the BLE notification length, some accept the negociation but don t respect it (Asus Zen Phone 2 for example).

gebner, 2021-03-04:
This happened on a Samsung Galaxy S10+, is there any debugging info I can gather to help?  Or any tweaks I can try out?

brvier, 2021-03-05:
The problem is that if there huge probability that m_tx_in_progress (https://github.com/daniel-thompson/micropython/commit/048bb2da83d0cbe73a829e461bd4f22b86c282bc) isn't reset to false in case of error. I ven t take a look at the softdevice used yet, but i ll try to debug that once i got my own device to test. With only a nrf52 dev kit, it ll be difficult to run the full wasp-os :)

brvier, 2021-03-17:
Also 
`while (m_tx_in_progress > MAX_TX_IN_PROGRESS) {`

Should be `
    while (m_tx_in_progress > = MAX_TX_IN_PROGRESS) {
`
Else we are off by one.
I'm current testing it on real device.

brvier, 2021-03-19:
24h later, no notifications received, gadgetbridge look like connected. So not related to this. Also step counter disapear of ring (related or not ...)

kj7rrv, 2023-08-17:
I haven't had the disconnection issue, but Wasp-os does miss quite a few notifications on my PineTime. This could also be related to #443.