wasp-os/wasp-os #148: Documentation updates

rgegriff, 2021-01-11:
Would you accept a pr adding a description of using nRF Toolbox to upload the bootloader/application? I have had a lot more success flashing using that (particularly if I limit the speed) over nRF Connect.

rgegriff, 2021-01-11:
I would also like to add `libnewlib-arm-none-eabi` as one of the build requirements, since I have been unable to build the firmware without that lately.

daniel-thompson, 2021-01-16:
> Would you accept a pr adding a description of using nRF Toolbox to upload the
> bootloader/application? I have had a lot more success flashing using that
> (particularly if I limit the speed) over nRF Connect.

No objections to improving the documentation but the details are important. There are already descriptions of nRF Connect so things will have to be slotted together carefully.

> I would also like to add libnewlib-arm-none-eabi as one of the build
> requirements, since I have been unable to build the firmware without
> that lately.

I'd be reluctant to do that. The CI builds do not require this package and the install guide deliberately recommends using the GNU-RM toolchain rather than using the one bundled with the distro (since the feature set of the bundled toolsets has, historically, been rather variable between distros).

rgegriff, 2021-01-17:
> No objections to improving the documentation but the details are important. There are already descriptions of nRF Connect so things will have to be slotted together carefully.

Agreed. The documentation as-is is really high-quality, and I think it will be important to try and match that, and make things flow to keep up to those standards. In that regard, if I was to take a crack at writing those instructions, do you think slotting it in between the nRF Connect and DaFlasher instructions would be the appropriate place? 

I was also thinking about adding a paragraph or at the top of the "Installing wasp-bootloader" and "Installing wasp-os" sections how there are three (I guess, four if you include nRF Toolbox) methods for flashing the bootloader and app onto the device.

I think the video does a lot of work supplementing the materials as well. What's your general feelings around including videos/screenshots to supplement the material as well?

> I'd be reluctant to do that. The CI builds do not require this package and the install guide deliberately recommends using the GNU-RM toolchain rather than using the one bundled with the distro (since the feature set of the bundled toolsets has, historically, been rather variable between distros).

Yea, I noticed that when I was reviewing the docs. It states that pretty clearly. I don't know how I missed it when I was building the dockerfile I am using as an environment. 🤦🏻 (and I should probably update my docker file to fetch the toolchain instead of using the distro libs so I don't run into any issues)

closed 2021-03-02 by rgegriff