wasp-os/wasp-os #155: Stuck on os, can't get in to DFU mode.

busbuyukefe, 2021-01-21:
I am running a newer P8 (magnetic charging). I installed the wasosp bootloader and micropython but I can't reboot to the bootloader by holding the power button. I tried connecting to it with the Serial Bluetooth Terminal app on android but it gives "MemoryError: memory allocation failed, allocating 512 bytes" on every command I type. Also when I press the "=" symbol on the calculator app it does nothing. How can I reboot to the bootloader mode?

leofidus, 2021-01-21:
I can only say that with the same watch (P8 with magnetic charging) I don't have this problem. When I hold the power button for five seconds I get the bootloader with the wasp logo. If that doesn't work and the terminal doesn't help then your best bet is to let it run out of power to force a reboot.

busbuyukefe, 2021-01-22:
Huh, I drained the battery and it seems like it is working as intended now.

closed 2021-01-30 by daniel-thompson