wasp-os/wasp-os #157: Pinetime unable to flash, OTA resets when upload starts

glaslos, 2021-01-22:
I successfully flashed `reloader-mcuboot.zip` from `wasp-os-0.3.tar.gz` to my Pinetime. When I try to flash `micropython.zip` with either DaFlasher or nRF Connect the watch switches to the pine cone + blue Bluetooth symbol screen once connected but the moment the upload starts, it switches back to the OTA start screen.

shuhaowu, 2021-02-02:
I have the same problem. When I try to go back to infinitime, the moment the upload starts, main.py starts...

glaslos, 2021-02-02:
@daniel-thompson I have an STLink coming my way. Any suggestions on how to debug this issue?

mycpuorg, 2021-02-04:
Same here. I was excited to try this out but I'm afraid that I might have bricked my sealed device :(
Please help @daniel-thompson 

[pine_error.log](https://github.com/daniel-thompson/wasp-os/files/5924101/pine_error.log)

The tail end of the log appears to show the error when the firmware update (Remote DFU) begins
```
D 00:01:35.233 [DFU] gatt.writeCharacteristic(00001531-1212-efde-1523-785feabcd123)
I 00:01:35.320 [DFU] Data written to 00001531-1212-efde-1523-785feabcd123, value (0x): 02-01
A 00:01:35.320 [DFU] Initialize DFU Parameters completed
I 00:01:35.320 [DFU] Notification received from 00001531-1212-efde-1523-785feabcd123, value (0x): 10-02-01
A 00:01:35.320 [DFU] Response received (Op Code = 2, Status = 1)
V 00:01:35.320 [DFU] Writing to characteristic 00001531-1212-efde-1523-785feabcd123
D 00:01:35.320 [DFU] gatt.writeCharacteristic(00001531-1212-efde-1523-785feabcd123)
I 00:01:35.410 [DFU] Data written to 00001531-1212-efde-1523-785feabcd123, value (0x): 03
A 00:01:35.411 [DFU] Receive Firmware Image request sent
A 00:01:35.428 [DFU] Uploading firmware...
V 00:01:35.428 [DFU] Sending firmware to characteristic 00001532-1212-efde-1523-785feabcd123...
I 00:01:35.499 [DFU] Notification received from 00001531-1212-efde-1523-785feabcd123, value (0x): 10-03-06
A 00:01:35.499 [DFU] Response received (Op Code = 3, Status = 6)
E 00:01:35.500 [DFU] Remote DFU error: OPERATION FAILED
V 00:01:35.500 [DFU] Writing to characteristic 00001531-1212-efde-1523-785feabcd123
D 00:01:35.500 [DFU] gatt.writeCharacteristic(00001531-1212-efde-1523-785feabcd123)

```

Please help! Thanks.

glaslos, 2021-02-04:
Pinetime improved Bluetooth stability by updating the BLE driver: https://github.com/JF002/Pinetime/releases/tag/0.12.0

daniel-thompson, 2021-02-04:
First try rebooting the phone, if that doesn't work you could try tweaking the DFU options in nRF Connect (reduce the number of packets and the packet size), if that doesn't work you'll need to try another device. If you haven't got alternative phones to try then there's a programmer for Linux hosts in tools/ota-dfu which works well on some laptop (and I'm not 100% sure but I think ota-dfu works with RPi4 as well).

glaslos, 2021-02-04:
Now I'm feeling a bit stupid for not trying it before :) Reducing the packet size to 2048 and keeping the number of packets at 10 did the trick for me. Thanks for the nudge!

mycpuorg, 2021-02-06:
@daniel-thompson @glaslos Thank you both for the suggestions. 
In addition to your suggestion I also had to toggle `Reliable Write` option which would have never occurred to me.

For future (BT n00bs like me):
+ If you face the same problem described in the main issue here, then reduce DFU packet size.
+ I am using nRF Connect app on my Pixel 3 phone.
+ You can do this on `nRF Connect App -> Settings -> DFU Options -> MBR size`. I set this to `1024` which worked.
+ For good measure I also reduced the `nRF Connect App -> Settings -> DFU Options -> Number of Packets` to `5` (from default 10)
+ Then I tried flashing DFU firmware but got an error "GATT in wrong state"
+ This error is not related to Nordic BT or Wasp OS AFAIK
+ This is an Android BLE device driver issue. I found this [SO question](https://stackoverflow.com/questions/61101913/android-ble-problems-writing-data-to-characteristic) helpful which recommends to turn off `Reliable Write`
+ But in my app I saw it was already turned off. Like a savvy technologist would do, I toggled this options (`BEGIN` -> `ABORT`)
+ Voila! my next attempt to flash Wasp OS succeeded!

mycpuorg, 2021-02-06:
@glaslos I recommend closing this issue as `Fixed`. Thanks.

closed 2021-02-06 by glaslos