wasp-os/wasp-os #159: pygobject and pydbus module are required for wasptool 'rtc' option

pstarliu, 2021-01-28:
Hi

I am following the document for flashing wasp-os into my pinetime, and it works the first time. 

But to set the initial time, both wasptool 'console' option and 'rtc' options doesn't working for me. I am not a python developer, but inspecting the error log that my guessing is 'pydbus' and 'pygobject' modules missing, and after install both, the command 'wasptool --rtc' successful set the correct local time. 

I am using a Pinebook pro with the preinstalled Manjaro Linux.

closed 2021-01-30 by daniel-thompson

pstarliu, 2021-01-30:
Hi @daniel-thompson, why it the issue been closed?

daniel-thompson, 2021-01-30:
github closes issues automatically when a fix committed.

pstarliu, 2021-01-30:
Cool, now I saw the linked Commit id, thanks @daniel-thompson

commit a1b1502d (2021-01-30): install: Include the wasptool dependencies in the install guide

Fixes: #159