wasp-os/wasp-os #160: Colors broken on P8

begmanii, 2021-01-29:
I compiled the latest git pull for my P8 and note some colors are broken.  In the self test, only the red slider is working.  The other 2 sliders do not work.

So 2048 is just and off-white rectangle and the YES NO boxes on the clear notifications dialog are also just off-white rectangles.

The Software Icon used to have colors, now it is yellow and white
[image]
[image]
[image]
[image]
[image]
[image]
[image]

begmanii, 2021-02-05:
I rebuilt again with latest pull and same thing.  I reset back to ATCWatch  and Softdevice 5.0.1 and reloaded from scratch. I still have the issue with the colors.

Am I the only one seeing this issue on the Colmi P8a?

daniel-thompson, 2021-02-06:
> Am I the only one seeing this issue on the Colmi P8a?

Nobody else has reported anything like this and I certainly don't see it on my own P8a. The ST7789 display controller does have some lookup tables that can shift the colour response but, given the first thing wasp-os does is reset the display then these should not be active. You could perhaps take a look at the ST7789 datasheet and fire off a couple of commands to try and change the digital gamma settings but I wouldn't be too optimistic, especially if you see similar behaviour using ATCwatch, since there's a possibility there is a hardware issue.

A really easy thing to try would be to just let the battery run down and see it the problem remains... again this is a long shot but at least it is easy.

begmanii, 2021-02-07:
I did not run ATCwatch, just reset using the ATCwatch bootloader.

I will try the slow death and see if that resets it.

Also, the button long press is not putting it into OTA mode either.

daniel-thompson, 2021-02-07:
The only thing likely to prevent long button press from working is installing the wrong binary (build-pinetime/micropython.zip for example).

begmanii, 2021-02-07:
I only build for the p8 as I do not have the PineTime.  Not sure why this is happening.

Can you build the p8 micropython.zip and send it to me so i can try that?  Maybe something is wrong with my machine or environment.

I really should have moved the old version to an archive so I can go back and test.  Lesson learned!

begmanii, 2021-02-07:
Funny thing is the button test works fine, just the long press to OTA does not....

busbuyukefe, 2021-02-07:
> Funny thing is the button test works fine, just the long press to OTA does not....

I had the same issue when I tried the wasp-os for the first time. Just turn on the flashlight and wait until the battery drains out, mine was fixed after a "force reboot"  :)

begmanii, 2021-02-08:
> > Funny thing is the button test works fine, just the long press to OTA does not....
> 
> I had the same issue when I tried the wasp-os for the first time. Just turn on the flashlight and wait until the battery drains out, mine was fixed after a "force reboot" :)

Thanks, At least that worked. Button operation is back

Now to fix the colors...

begmanii, 2021-02-08:
Well, I bricked it.  I tried the DS-D6-adafruit and ATCdfu again to reset.  However, I used my Linux box to flash the bootloader-daflasher.zip and now I have a black screen with no logo.  BLE scans via PC, DaFlasher app and NRF Connect no longer show a functional BT

Sads...   :(

begmanii, 2021-02-23:
I purchased a STLink programmer, but can't program the bootloader.hex file.  It seems the chip is locked?

I can't get the init to work under OpenOCD.  I keep getting the same error:

Error: init mode failed (unable to connect to the target)

Research on the web is either I have the wrong pins connected or the NRF chip is locked  (tried every combination with guidance from Aaron's videos) and no dice.  I do not have a Jlink to try the unlock, don't know if I want to spend that money if it does not work!

@daniel-thompson Do you have any suggestions?

SteveAmor, 2021-02-23:
@busbuyukefe it's unlikely that the NRF is locked as the DaFlasher processes removes the flash protection.  Sorry I can't offer much more than that.

begmanii, 2021-02-23:
Thanks @SteveAmor That is what i thought as well.  Arron has a very good video where he describes the 3.3V, GND, DIO and CLK pins.  I can confirm the 3.3/Gnd work as the watch powers the STLink, at least the STLink LED turns on and flashes when connected to watch only, not USB.

I tried changing the CLK and DIO pins around and still no dice.

begmanii, 2021-02-23:
Even built openocd from scratch.

[image]

begmanii, 2021-03-05:
Well, I was able to get a Segger J-Link programmer and much the same issue.  The programmer was not able to connect to the device to do anything at all (erase, program or just initialization).

I think the chip is dead, although I have no idea what happened as it downloaded the software correctly and did not come back up after reboot.

Thanks all for your help.

closed 2021-06-03 by daniel-thompson