wasp-os/wasp-os #162: Unable to connect to gadget bridge

shuhaowu, 2021-02-02:
I just built wasp-os on master and flashed it onto the pinetime. I'm unable to find it in Gadget bridge. I can see it in nrf  connect, but I only see a Nordic UART service and two Generic attribute (uuid 1800 and 1801). Are features like the time service, notification, and what not simply not implemented, or am I doing something wrong?

Kozova1, 2021-02-02:
Are you using the [custom gadgetbridge fork](https://github.com/daniel-thompson/gadgetbridge)?

shuhaowu, 2021-02-03:
I am not. I see [upstream is looking to resolve some conflicts](https://codeberg.org/daniel-thompson/Gadgetbridge/pulls/1). Does anyone own this work? If not, I can try my hands rebasing that change and sending it upstream.

Kozova1, 2021-02-06:
Upstream has merged wasp-os, now we either need to wait for a new release or build from source.

Kozova1, 2021-02-16:
@shuhaowu Gadgetbridge just released a new version that supports wasp-os out of the box - 0.53

closed 2021-02-20 by daniel-thompson