wasp-os/wasp-os #165: Improve OTA Flash Speed

Kozova1, 2021-02-06:
Currently, when using nRF Connect on my android phone to flash, it flashes at 2-3 kbps.  
This is even worse with wasptool, which flashes at an excruciatingly slow 0.3 kbps.

With Infinitime, I get speeds of ~25-35 kbps, which is WAY better.

Carglglz, 2021-02-06:
The speed difference is likely to be affected by MTU size.

The MTU size of the [ boot loader](https://github.com/daniel-thompson/wasp-bootloader/blob/1d128f1311c084c23a662cb68263df9b187ebf94/src/main.c#L117) is 247 bytes and the MicroPython nrf port [BLE stack](https://github.com/daniel-thompson/micropython/blob/048bb2da83d0cbe73a829e461bd4f22b86c282bc/ports/nrf/drivers/bluetooth/ble_drv.c#L77) is 23 bytes.

So increasing the MTU size is probably the way to solve this, although it may require a pine time with SWD, since this increase needs some sort of RAM reallocation. [Increasing MTU size](https://devzone.nordicsemi.com/f/nordic-q-a/56441/increasing-mtu-size) 👀

rgegriff, 2021-03-18:
I am not convinced of this. I am able to get MUCH, MUCH faster flashes using nRF Toolbox (not Connect) in the DFU mode. You need to go into the DFU settings and set "Number of Packets" to 10. (the default settings send way too fast and cause failures.

Flashing the firmware this way I am able to do the entire micropython.zip in under a minute. This implies that there are things that could be done to improve the speed of wasptool.

fanoush, 2021-04-26:
hi, found this github issue and as I was interested in this some time ago here are  few random comments:

speed is about MTU, DLE but also about connection interval, and number of packets sent in one connection interval (i.e. even with small MTU it can be fast when multiple packets are sent in one interval so the interval is not wasted and/or the interval is shorter)
some good info 
https://punchthrough.com/maximizing-ble-throughput-on-ios-and-android/
https://www.novelbits.io/bluetooth-5-speed-maximum-throughput/
also it is about complexity of communication e.g. the packet receipt notification in DFU protocol is slowing it down and is optional and can be turned off completely to speed it up a bit (the number of packets 10 you mentioned, it can be turned off instead) 

some old discussion http://forum.espruino.com/conversations/338789/?offset=25#14953321
also looks like we are using same python OTA flashing code based on https://github.com/dingari/ota-dfu-python , some time ago I have added MTU negotiation to it and it helped - check this python script https://github.com/fanoush/ds-d6/blob/master/fwdump/desay_dfu.py maybe it will work out of box with waspos device too? I did not play with connection interval at that time, maybe that would improve it too.

as for adding larger MTU to micropython, this is unrelated to DFU speed but it is indeed possible and relatively easy, recently I managed to add it to Espruino for nrf52 and it seems to work and the C code itself is quite short, see https://github.com/espruino/Espruino/pull/1998/files 
The " sort of RAM reallocation" is about linker script and indeed softdevice does not start and the device becomes inaccessible if you set it too low, but if you are careful and set it higher it will start and then you can check how much it really needed and scale it down next time - the `nrf_sdh_ble_enable(&app_ram_base)` checks it and then fills correct value, in Espruino pull request I have listed some APP_RAM_BASE values for some MTU values