wasp-os/wasp-os #166: MemoryError when adding Game of Life using Software app when all other apps are enabled.

Kozova1, 2021-02-12:
Steps to reproduce (on a PineTime):  
1. Open software app
2. Enable all apps excluding Game of Life
3. Enable Game of Life
4. Out of memory error

Reproduced on 2 PineTimes.
It also seems to happen sometimes when adding Game of Life, even without adding all apps, but I cannot reproduce this reliably.

daniel-thompson, 2021-02-17:
I have two answers for this (depending on which one you like best).

1. I can't reproduce this. Do you have any custom code in your main.py that is absorbing memory? There has been some modest memory tuning in the last month but it shouldn't be enough to make the difference on this.... for me the test app is showing >2k free with all apps enabled.

2. The purpose of the Software app is to allow us to have more apps on the watch than can fit in memory. Even if I can't reproduce today there will certainly come a point when it is not possible to enable all the apps at the same time. So I'm ok with the out-of-memory message... but what happens after that (can you dismiss it and continue using the watch or does it form a loop that you have to break with a reset?).

Kozova1, 2021-02-17:
The watch keeps working after the MemoryError.  

IMO the message could be prettified a bit, instead of a crash.

daniel-thompson, 2021-06-20:
I believe this was fixed by #199.

closed 2021-06-20 by daniel-thompson