wasp-os/wasp-os #167: OverflowError when using one of the 'spinner'-widgets in the 'Date'-settings on the PineTime

christopherpeters-git, 2021-02-16:
Steps to reproduce:
1. Open 'settings' app
2. Swipe to 'Date'
3. Tap two times on one of the spinners

After the display of the exception a chain of other exceptions of the same type occure and i can only restart it to use it again.

The error is raised by 'time.localtime(...)' in 'get_localtime()' in apps/nrf_rtc.py. 'time.localtime(...) seems to raise the error when the input is greater than or equals 1 << 31, like a guy states in the [micropython forum](https://forum.micropython.org/viewtopic.php?t=8486)

I cant reproduce this in the simulator.

closed 2021-02-20 by daniel-thompson

christopherpeters-git, 2021-02-21:
May i ask, why this issue was closed? This was my first issue i ever created in foreign project, so i would like to know what the reason is. @daniel-thompson

daniel-thompson, 2021-02-21:
github automatically closes issues when they are fixed. Follow the link in "closed this" message above!

christopherpeters-git, 2021-02-21:
sorry, didnt knew that! Thank you for the fix

commit 090ed848 (2021-02-20): apps: settings: Clamp the dates a little more tightly

Currently a couple of down presses on the year results in the time being
set to some time in 2099. The micropython date logic does not support
dates this far in the future and throws an exception. Adopt a simple
fix that should be OK for the next 39 years ;-) .

Reported-by: Christopher Peters <me@christopherp.de>
Fixes: #167