wasp-os/wasp-os #176: support P8 CST716 touch driver

ildar, 2021-03-04:
I installed the FW on my P8 and found that touch doesn't respond. It probably has another touch IC. There's a video on that topic: https://youtu.be/jgjKaSETY8Y
ATCwatch FW worked. So I guess that this would help: https://github.com/atc1441/ATCwatch/commit/7f693d9ebed80acf56723eff75c51ac648d7829e

daniel-thompson, 2021-03-04:
wasp-os is already tested on both known versions of the touch screen device (and does not use the X/Y data on swipe so it does not need a fix like the one seen in ATCwatch).

pebri86, 2021-03-05:
Maybe something wrong when ts chip initialization, it happen once to my P8. Let the battery drain out then it will work.

ildar, 2021-03-05:
100% right. Power drain cycle helped.

closed 2021-03-05 by ildar