wasp-os/wasp-os #185: Motion Sensor axes possibly mapped incorrectly

rgegriff, 2021-03-16:
Hi!

I've been hacking on the motion sensor trying to get a pin to trigger with the WRIST_WEAR feature. While working on this (and banging my head against the BMA documentation) I added a function to the BMA421 class to let me read the raw accelerometer data:
```
    def read_accel_xyz(self):
        return bma42x.BMA42X.read_accel_xyz(self._dev)
```
If I am understanding the returned tuple correctly, I think the X and Y axes are flipped. Z seems correct.
[image]

When it's resting on the table, Z shows a very high magnitude negative result (expected). 
`(22, 157, -1036)`
This is assuming the tuple is in the form (x, y, z).

However, when I tilt it to the left, I get a similarly high negative reading on the Y axis.
`(40, -795, -451)`

And when I tilt it with the bottom toward me, I get a negative X reading.
`(-1004, 143, -117)`

Is this incorrect? Or am I misinterpreting the expected output from `bma42x.BMA42x.read_accel_xyz()`?

daniel-thompson, 2021-03-17:
That sounds familiar. Are you sure the BMA421 is mounted in the chassis the way you have diagrammed it?

rgegriff, 2021-03-18:
I don't believe it is. I disassembled the devkit watch I have but was unable to 100% identify which chip the BMA421 is; and I was unable to find any high res images of the internals. My guess is it's the one mounted to the orange pfc cable under the battery, but that's just a guess. I can't read the markings on it with my eyes.

That diagram is just from the BMA423 application notes on the chip features. (which doesn't seem to be on the bosch website any more, but if anyone needs it, feel free to email me). I just included it for reference for the expected orientation of the axes to get correct readings and for the "smart" features (wrist wear wakeup being the most interesting to me, personally. But I think the activity classification could also be really useful for distance traveled estimation).

I saw there is some c code for remapping the axes, but I don't think it's exposed on the python side. I just wanted to make sure that I am interpreting the output of `read_accel_xyz` correctly.

If so, I will learn how the c modules work to expose that function to python so that I can remap the axes when we start the chip in `BMA421.reset()`

rgegriff, 2021-03-18:
Thinking about this some more, I think the Z axis' sign is inverted as well. If the watch is face down, gravity would be pulling toward the "top" of the chip.
So I think the correct remapping would be: swap X-Y axis, flip sign of the Z axis.

rgegriff, 2021-03-18:
Wow... I feel foolish. I finally found some documentation for the BMA421 and it doesn't seem to have wrist wear or double-tap detection 🤦🏻

The only features listed are "step counter" and "motion/no motion"

Well, I can't say I /wasted/ my last few evenings of free time since I learned some stuff, but I certainly could have spent them more productively, haha.

In light of this, unless someone can figure out how to activate these features (I wouldn't be surprised if they were using the same silicon) it's probably not remapping the axes on the chip itself. Might just be easier to remap them in python. I'll open a PR with that available from watch.accel.read_xyz()

Closing this.

closed 2021-03-18 by rgegriff