wasp-os/wasp-os #192: draw565.py: blit Image format not detected correctly

bmsleight, 2021-03-21:
See https://gist.github.com/bmsleight/8785aa85aecea33037f2ff54abaf5a89
```python
    def _draw(self):
        """Draw the next frame."""
        draw = wasp.watch.drawable
        
        # Works with micropython logo
        self._logo = micropython
        draw.rleblit(self._logo)

        # Works with calling _rle2bit directly
        self._logo = demo_icon  
        draw._rle2bit(self._logo, 0, 0, fg=0xffff, c1=0x4a69, c2=0x7bef)
        
        # Does not work with demo logo & rleblit
        self._logo = demo_icon  
        draw.rleblit(self._logo)
        # Crashes on above line
```

```python

>>> wasp.system.run()
Watch is running, use Ctrl-C to stop

Traceback (most recent call last):
  File "/home/bms/wasp-os/wasp/wasp.py", line 497, in run
    self._tick()
  File "/home/bms/wasp-os/wasp/wasp.py", line 459, in _tick
    self._handle_touch(event)
  File "/home/bms/wasp-os/wasp/wasp.py", line 419, in _handle_touch
    self.app.touch(event)
  File "/home/bms/wasp-os/wasp/apps/launcher.py", line 55, in touch
    wasp.system.switch(app)
  File "/home/bms/wasp-os/wasp/wasp.py", line 250, in switch
    app.foreground()
  File "/home/bms/wasp-os/wasp/boards/simulator/demo.py", line 155, in foreground
    self._draw()
  File "/home/bms/wasp-os/wasp/boards/simulator/demo.py", line 177, in _draw
    draw.rleblit(self._logo)
  File "/home/bms/wasp-os/wasp/draw565.py", line 195, in rleblit
    (sx, sy, rle) = image
ValueError: too many values to unpack (expected 3)
```

daniel-thompson, 2021-03-21:
`rleblit()` only supports 1-bit images. I you want automatic format detection use `blit()`.

bmsleight, 2021-03-21:
I should read the code better = sorry

closed 2021-03-21 by bmsleight