wasp-os/wasp-os #193: draw565.wrap create too large index if str end with space

brvier, 2021-03-22:
Given the following string with font24:
```
draw.set_font(fonts.sans24)
s = "'A Name really long which break'"
```
draw565.wrap(s,width) return [0, 19, ...] instead of [0, 18].

What is the purpose of testing space here in the string ?
https://github.com/daniel-thompson/wasp-os/blob/86cc4844b6cb8278789971d9fb0afdb710e8d253/wasp/draw565.py#L376

daniel-thompson, 2021-03-22:
> draw565.wrap(s,width) return [0, 19, ...] instead of [0, 18].

The wrap() algorithm currently breaks after spaces (since having a space character on the left margin looks bad). That means the correct result for this would be [0, 15, ...] .

> What is the purpose of testing space here in the string ?

It is remembering the best place to break a line (so we can prefer to break on word boundaries).

You're running mostly on the simulator? I think the recent changes to the simulator to prevent drawing off-the-screen will have picked up this as a bug (which it is). Does the PR above fix the problem?

I added a couple of comments to better show how the loop works.

brvier, 2021-03-24:
Indeed, i did think a soft wrap :)
This is now far better.
:+1:

commit 402801c5 (2021-03-27): draw565: Avoid over-long lines when handling space

Currently, if the line wrapper attempts to break a line on a space and
that space character is outside the bounding box, then we generate an
over-long line.

Fix this by handling line break generation *after* we have checked the
length of the current line.

Fixes: #193
PR #194 (2021-03-27): draw565: Avoid over-long lines when handling space
Currently, if the line wrapper attempts to break a line on a space and
that space character is outside the bounding box, then we generate an
over-long line.

Fix this by handling line break generation *after* we have checked the
length of the current line.

Fixes: #193
closed 2021-03-27 by daniel-thompson