wasp-os/wasp-os #205: Can't find binary release of reloader-factory

doniks, 2021-05-05:
In the description of this video https://www.youtube.com/watch?v=lPasAt1LJmo
there is a link to https://github.com/daniel-thompson/wasp-os/actions/runs/350930611
where I should be able to get the binaries. But I can't see them. I'm not too familiar with github actions, but maybe there were some CI artifacts here once but they expired and were removed!?

daniel-thompson, 2021-05-05:
Yes, I think they have timed out. Check out the [install guide](https://wasp-os.readthedocs.io/en/latest/install.html) for info on where to get binary releases.

doniks, 2021-05-05:
ah, ok. now I see. The release contains one zip file, inside is a folder called build-pinetime and inside are the two reloader-*.zip files. 

Ok. I could imagine that other users might also struggle to find this. But, up to you of course. My itch has been scratched. I'm closing it.

closed 2021-05-05 by doniks