wasp-os/wasp-os #206: Colmi P8 blackscreen after bootloader install

razem-io, 2021-05-07:
Hi,

first thank you for this project. I might be able to enjoy smartwatches again 👍 ! Sadly, I have a bad start.

I followed this tutorial https://www.youtube.com/watch?v=VJoDtMy-4pk and everything looked fine until ~2:28, after "Done, restarting". The smartwatch is completly black and I do not see it anymore in DaFlasher (also not the mac address).

What I tried so far:
* pressing the button
* holding the button for 10 sec
* holding the button for 30 sec

Is it bricked or are there other solutuions? I checked the troubleshooting section and some of the issues here but did not find anything related to my problem.

daniel-thompson, 2021-05-07:
Really sorry to year this and I'm afraid it is certainly possible that the system is soft-bricked. Aaron added a warning to the DaFlasherFiles repo:
"ATTENTION!!!! DO NOT FLASH ANY MOY-TON5 devices right now it will brick the watch. See the model model infos in the Watch menu"

I'm still trying to double check but I think the MOY-TON5 might be a new (undocumented) update to the P8 family. As a precaution I've added a banner warning to the description in the video but that might be too late to help you.

There may be other solutions but they are pretty tricky to pull off and generally need quite a lot of microcontroller experience (use heat to disassemble the watch, find the four pin programming header and reprogram the device with a dongle).

razem-io, 2021-05-07:
Thank you @daniel-thompson for you fast response. I guess I will try another batch and maybe later watch a few videos on how to open my bricked one and fix it. Closing, as this is not an issue with wasp-os.

daniel-thompson, 2021-05-07:
I'm sorry you had a difficult start. If it helps my own start with smartwatch hacking wasn't all that different. I didn't actually brick mine: I took it appart to solder in some extra wires to make it hackable and my soldering simply wasn't up to it meaning I physically destroyed the device instead.

PineTime is worth a look (for all the reasons [mentioned in the wasp-os docs](https://wasp-os.readthedocs.io/en/latest/install.html#device-support) ). Whilst I have since branched out and experimented with several other devices (mostly for cosmetic variety). However I don't think wasp-os would have happened without PineTime boosting *my* confidence first!

closed 2021-05-08 by daniel-thompson