wasp-os/wasp-os #207: BLE GATT interface?

jones139, 2021-05-08:
[this is probably not an issue, but me failing to understand something, sorry!]
I am struggling to find the bluetooth module on the device.    I would like to implement something like the micropython ble_temperature.py example (but for accelerometer and heart rate data that I need for my OpenSeizureDetector project).
It looks like micropython has been compliled with an SD=s132 parameter which I think should have enabled bluetooth on the device - but I can't do 'import bluetooth' and cant see bluetooth under the micropython or wasp modules.
Can anyone point me in the right direction please?
Thanks
Graham.

jones139, 2021-05-08:
ah....I think it might be called ubluepy, not bluetooth, will check and close this issue if that is the case....

daniel-thompson, 2021-05-08:
Yes. That's about the size of it. The micropython ports for different microcontrollers don't offer identical APIs hence the nRF port uses ubluepy to get access to low-level BLE APIs.

jones139, 2021-05-08:
Thanks!  Nearly working - just need to resolve the advertising problem from issue #110.
Graham.

jones139, 2021-05-13:
[Sorry to re-open this one, but I don't seem to be able to re-open #110]

I can not get advertising working now i have found ubluepy and some useful examples that use it.   The last comment on #110 was:

> This is because with NUS service enabled, the device is already advertising at that point, so you need to stop it first and then advertise again

I have not worked out how to stop advertising the NUS service and re-start it alongside other services.  Calling (from python) ubluepy.Peripheral().advertise_stop() seems to work, but the advertise command still gives me the same error - so I suspect that the NUS service is running on a different instance of Peripheral?.

I have found that in micrpython/ports/nrf/main.c there is a call to ble_uart_init0(), which is defined in micropython/ports/nrf/drivers/bluetooth/ble_uart.c. ble_uart_init0() starts the advertising (from C).

Is the best way to inhibit this starting in main.c and try to start the ble uart from python along with my services - or is there another way? (sorry, I am not used to micropython!).   
I would like to use micropython and wasp-os, given all the work that has gone into it, but i need to be able to set up new BLE GATT services - my fall back option is to use a lot of the micropython C code and write the app in C rather than python, but would like to use micropython properly if anyone knows how to do it.

Thanks

Graham.

Pittconnect, 2022-02-01:
@jones139 Can you please update us at https://github.com/daniel-thompson/wasp-os/issues/299 I am not sure if this is a duplicate issue or a different problem altogether

jones139, 2022-02-01:
@Pittconnect , @rennard is working on the issue that I failed to resolve.   I'd suggest that once he has sorted his issue with characteristic uuids that we put a little example in this issue for anyone looking in the future and close it.

jonnor, 2024-12-27:
@jones139 @Pittconnect is this still relevant? Or has it been resolved?

jones139, 2024-12-27:
I don't know, I am afraid.   I used a modified version of infinitime
instead.  Regards,  Graham