wasp-os/wasp-os #209: -Werror=maybe-uninitialized is triggered when I try to build the bootloader for the pinetime.

Liorst4, 2021-05-15:
```
make BOARD=pinetime all
rm -f bootloader/_build-pinetime_nrf52832//pinetime_nrf52832_bootloader-*-nosd.hex
make -C bootloader/ BOARD=pinetime_nrf52832 all genhex
make[1]: Entering directory 'wasp-os/bootloader'
CC main.c
src/main.c: In function 'proc_ble':
src/main.c:421:24: error: '*(ble_evt_t *)(&ev_buf[0]).header.evt_id' may be used uninitialized [-Werror=maybe-uninitialized]
  421 |     switch (evt->header.evt_id)
      |             ~~~~~~~~~~~^~~~~~~
src/main.c:410:22: note: 'ev_buf' declared here
  410 |   __ALIGN(4) uint8_t ev_buf[ BLE_EVT_LEN_MAX(BLEGATT_ATT_MTU_MAX) ];
      |                      ^~~~~~
src/main.c: In function 'proc_soc':
src/main.c:452:5: error: 'soc_evt' may be used uninitialized [-Werror=maybe-uninitialized]
  452 |     pstorage_sys_event_handler(soc_evt);
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/main.c:447:12: note: 'soc_evt' declared here
  447 |   uint32_t soc_evt;
      |            ^~~~~~~
cc1: all warnings being treated as errors
make[1]: *** [Makefile:355: _build-pinetime_nrf52832/main.o] Error 1
make[1]: Leaving directory 'wasp-os/bootloader'
make: *** [Makefile:38: bootloader] Error 2
```

A simple workaround is
```diff
--- a/src/main.c
+++ b/src/main.c
@@ -407,7 +407,7 @@ void assert_nrf_callback (uint16_t line_num, uint8_t const * p_file_name)
 // Process BLE event from SD
 uint32_t proc_ble(void)
 {
-  __ALIGN(4) uint8_t ev_buf[ BLE_EVT_LEN_MAX(BLEGATT_ATT_MTU_MAX) ];
+  __ALIGN(4) uint8_t ev_buf[ BLE_EVT_LEN_MAX(BLEGATT_ATT_MTU_MAX) ] = {0};
   uint16_t ev_len = BLE_EVT_LEN_MAX(BLEGATT_ATT_MTU_MAX);
 
   // Get BLE Event
@@ -444,7 +444,7 @@ uint32_t proc_ble(void)
 // process SOC event from SD
 uint32_t proc_soc(void)
 {
-  uint32_t soc_evt;
+  uint32_t soc_evt = 0;
   uint32_t err = sd_evt_get(&soc_evt);
 
   if (NRF_SUCCESS == err)
```

Liorst4, 2021-05-15:
BTW im using gcc-11

sethmiller, 2021-07-01:
I get the same issue. You can workaround by patching the bootlader makefile to ignore the error (ignoring errors is probably not a good permanent fix though):

```patch
diff --git a/Makefile b/Makefile
index bca5edf..e0c0de8 100644
--- a/Makefile
+++ b/Makefile
@@ -231,6 +231,7 @@ CFLAGS += -mcpu=cortex-m4
 CFLAGS += -mthumb -mabi=aapcs --std=gnu99
 CFLAGS += -Wall -Werror -Os -g3
 CFLAGS += -mfloat-abi=hard -mfpu=fpv4-sp-d16
+CFLAGS += -Wno-maybe-uninitialized
 
 # keep every function in separate section. This will allow linker to dump unused functions
 CFLAGS += -ffunction-sections -fdata-sections -fno-strict-aliasing
```

daniel-thompson, 2021-08-01:
Does the bootloader actualy work after these changes? There are reports that the gcc-11 from ArchLinux compiles ok but results in a bootloader that boot loops continually (which is enough to break sealed units).. and I was able to reproduce when i tried the same thing

sethmiller, 2021-08-02:
That's what happened to me. I had assumed I had done something else wrong and have not had time to try to get it back to a good state (thankfully it was an open unit so I should be able to).

Any pointers on getting it up and running again? Is there I guide I can run through for flashing via a physical connection?

daniel-thompson, 2021-08-13:
> Any pointers on getting it up and running again? Is there I guide I can run through for flashing via a physical connection?

See https://wiki.pine64.org/wiki/Reprogramming_the_PineTime

daniel-thompson, 2021-09-05:
So... I think I was right about the cause of the probelm but the above workaround doesn't quite work so, for now, I have added a different one.

As a side effect this also fixes the warnings without code changes.

closed 2021-09-05 by daniel-thompson