wasp-os/wasp-os #211: Python 3.7 and 3.9 build error

engenegr, 2021-05-20:
While building project I got error

```
python3 -m nordicsemi dfu genpkg \
	--dev-type 0x0052 \
	--application micropython/ports/nrf/build-p8-s132/firmware.hex \
	build-p8/micropython.zip
Traceback (most recent call last):
  File "/home/ilya/.local/python/3.7.9/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/home/ilya/.local/python/3.7.9/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/ilya/Code/Python/wasp-os/tools/nrfutil/nordicsemi/__main__.py", line 315, in <module>
    cli()
  File "/home/ilya/Code/Python/wasp-os/env/lib/python3.7/site-packages/click/core.py", line 1137, in __call__
    return self.main(*args, **kwargs)
  File "/home/ilya/Code/Python/wasp-os/env/lib/python3.7/site-packages/click/core.py", line 1062, in main
    rv = self.invoke(ctx)
  File "/home/ilya/Code/Python/wasp-os/env/lib/python3.7/site-packages/click/core.py", line 1668, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/ilya/Code/Python/wasp-os/env/lib/python3.7/site-packages/click/core.py", line 1666, in invoke
    sub_ctx = cmd.make_context(cmd_name, args, parent=ctx)
  File "/home/ilya/Code/Python/wasp-os/env/lib/python3.7/site-packages/click/core.py", line 923, in make_context
    self.parse_args(ctx, args)
  File "/home/ilya/Code/Python/wasp-os/env/lib/python3.7/site-packages/click/core.py", line 1379, in parse_args
    value, args = param.handle_parse_result(ctx, opts, args)
  File "/home/ilya/Code/Python/wasp-os/env/lib/python3.7/site-packages/click/core.py", line 2364, in handle_parse_result
    value = self.process_value(ctx, value)
  File "/home/ilya/Code/Python/wasp-os/env/lib/python3.7/site-packages/click/core.py", line 2320, in process_value
    value = self.type_cast_value(ctx, value)
  File "/home/ilya/Code/Python/wasp-os/env/lib/python3.7/site-packages/click/core.py", line 2307, in type_cast_value
    return convert(value)
  File "/home/ilya/Code/Python/wasp-os/env/lib/python3.7/site-packages/click/types.py", line 75, in __call__
    return self.convert(value, param, ctx)
  File "/home/ilya/Code/Python/wasp-os/tools/nrfutil/nordicsemi/__main__.py", line 64, in convert
    if value.lower() == 'none':
AttributeError: 'int' object has no attribute 'lower'
```
which I fixed quickly by

```
        try:
            if value.lower() == 'none':
                return 'none'
            return int_as_text_to_int(value)
        except AttributeError:
            return value
```

Warning! Not tested yet.

daniel-thompson, 2021-06-04:
What distro are you using, what version of wasp-os and how did you install the prerequisite packages?

I do not see anything like this on any platform I have tested on (including both python 3.7 and python 3.9).

NLM52679, 2021-06-12:
i get this on postmarketos on my pbp which afaik uses alpine's repos for everything followed the readthedocs build instructions, had to install bash for some scripts that require it https://pastebin.com/wDbrkUsC (my list of python packages and toolchain used)

engenegr, 2021-06-12:
@daniel-thompson it was Python 3.7 built in Debian Bullseye.

As far as I remember I installed packages via `pip install -r requirements.txt`

commit 29d619ae (2021-06-20): tools: Update nrfutil for click 8.x compatibility

Tested on Debian bullseye using wasp/requirements.txt in order to get
the latest version of click.

Fixes: #211
closed 2021-06-20 by daniel-thompson