wasp-os/wasp-os #214: Unregistering apps doesn't remove the module containing the app code, just the single instance of the application.

Liorst4, 2021-05-22:
Steps to re-create:

1. Reboot the watch
2. Connect to console
3. from the console ```python
import sys
sys.modules
```
4. Go to the software app and mark 2048
5. from the console ```python
>>> sys.modules
>>> wasp,gc.mem_free()
```
2048 will be there.
6. Go to software app and uncheck 2048
7. from the console ```python
>>> sys.modules
>>> wasp.gc.mem_free()
````
2048 will still be there and the GC didn't collect the assets from the 2048 module.

closed 2021-06-03 by daniel-thompson

Liorst4, 2021-06-05:
https://github.com/daniel-thompson/wasp-os/commit/e76a4afd8550145d4c43b3f1513b2fd209af8dd1 Reloads the application, but `unregister` still doesn't delete the module, just the app instance.

daniel-thompson, 2021-06-05:
>   e76a4af  Reloads the application, but  unregister  still doesn't delete the module, just the app instance.

Thanks for the original report BTW. GitHub is a little abrupt when it auto-closes issues like this!

>  e76a4af Reloads the application, but
  
>  unregister still doesn't delete the module, just
  
>  the app instance.
  

  
There is no new code in the unregister because we can drop the references to the module from the register() function instead.

Liorst4, 2021-06-05:
> There is no new code in the unregister because we can drop the references to the module from the register() function instead.

Yup, realized it after I wrote the comment :P
Thought that deleting the comment will spare you the notification.

> Thanks for the original report BTW

Glad to help! :smile:

commit e76a4afd (2021-06-03): manager: Improve application registration

When an application is registered using a string that gives the class
name (e.g.  "apps.chrono.ChronoApp") when we automatically delete
the module from a couple of namespaces. This ensures the garbage
collector can do a deeper clean when the application is unregistered.

We also provide a means to directly register watch faces (e.g. to
replace the default clock).

Fixes: #214