wasp-os/wasp-os #215: Battery exhausting too quickly

mzdaniel, 2021-05-26:
Thank you so much for this project Daniel.

Its being like forever since I wanted to have a cool gadget I can customize based on micropython, and this project did it perfectly.

I'm finding that battery life is quite short, around little more than 1 day and feels like my first micropython trials where the device was constantly draining battery for unused resources. 

Not sure is it is posible to keep the timers in light sleep, but I suspect that bluetooth constantly advertising would be a major cause. I have already contributed to micropython nrf and happy to help and create PRs with some guidance

daniel-thompson, 2021-06-04:
That would be great.

I can't remember where this is written down most lucidly but, with wasp-os, I don't want to distract attention from the micropython upstream. Therefore I usually recommend submitting patches to micropython although I am willing to absorb useful patches from PRs if the upstream it slow to do the code review.

In this case what is needed is some additional APIs in ports/nrf/drivers/bluetooth/ble_uart.c to enable/disable advertizing. These can then be made python facing in the same way as APIs like ble_uart_connected() [1].

BTW I think battery life took a bit of a nose dive recently as we added more logging features for the step counter (which means we run more code in the period tick handler... which costs power). I've just been trying to optimize that code a little. Be aware that even if this works I doubt it will extend battery life much beyond three days though.

[1] I think this might not actually be upstreamed since it pre-dates the upstream-first policy I mentioned above!

kLabz, 2021-09-25:
> BTW I think battery life took a bit of a nose dive recently as we added more logging features for the step counter

I can confirm that. I disabled the step counter app on my pinetime, and after 24 hours, instead of losing pretty much a full battery like previous days, I lost 30-35% battery (from 72~77% to 43%).

thiswillbeyourgithub, 2021-12-29:
Hi,

With user @leofidus in issue #273 we were wondering what would be the expected battery life using wasp os compared to pinetime, given the fact that optimized C++ is more efficient than micro python.

I'm specifically asking about limitations inherent to the language and not to the current state of both projects (wasp os is younger and I expect battery life to improve in the coming months).

Thanks :) !

kLabz, 2021-12-29:
For what it's worth, I did 4 days and 18 hours on a pinetime with a modified wasp-os, but with low usage.

thiswillbeyourgithub, 2021-12-29:
Thanks for the answer. Can you also give me an idea of battery life with average usage?

Also, do you have any idea of the room for improvement? The main factor is probably micro python vs C++.

Pittconnect, 2021-12-29:
Hello @kLabz 18hrs seems to be about right with modified wasp-os and 12 to 14 hours without  any modification What modification did you do @kLabz ? Would be curious to get started.

kLabz, 2021-12-29:
Not 18h, but 4 days + 18 hours, so 114 hours. That was mostly with step counter disabled iirc.

thiswillbeyourgithub, 2021-12-30:
Thanks, and no idea of the theoretical upper limit user could expect? Is there a module implemented to get real time power consumption? That would be useful.