wasp-os/wasp-os #216: After git fetch origin cannot run make successfully

Pittconnect, 2021-06-10:
[image]

daniel-thompson, 2021-06-10:
Looks like your pyserial installation Is damaged. Why is there no Serial in your copy of /home/pi/.local/python3.7/site-packages/serial/__init__.py ? Perhaps uninstalling and reinstalling it could help.

Pittconnect, 2021-06-10:
I did try that and here is the video Did I miss anything ? https://www.youtube.com/watch?v=oN4jNTBleLY

Pittconnect, 2021-06-10:
Maybe I can manually copy serial but if I knew which one and where that would be helpful.

[image]

daniel-thompson, 2021-06-11:
The click module is included in the build instructions: https://wasp-os.readthedocs.io/en/latest/install.html#install-prerequisites

What technique did you use to install the click module?

Pittconnect, 2021-06-12:
Ok I did the pre-installed for wasp-os again as python3 was removed.  And I am back to the the message cannot import serial
Here is the video https://www.youtube.com/watch?v=YpQ4_siN8Cg for details And I am not using docker should I be using docker ?

daniel-thompson, 2021-06-12:
Please can you share copy 'n paste from the terminal emulator. I can't keep track of things in video form?

Pittconnect, 2021-06-12:
I believe it may be something to do with user / system permission and how the file are arranged on the system.
Here is the paste version of it. https://pastebin.pl/view/f05aa461  Notice How I get micropython Error 2 when I use sudo command to make the firmware and when I use the local user to make the firmware I get Error 1 and the 3rd I have displayed the content of init.py

Pittconnect, 2021-06-14:
OK I found the work around https://stackoverflow.com/questions/65788196/cannot-import-name-serial-from-serial

closed 2021-06-14 by Pittconnect