wasp-os/wasp-os #218: How can i write an application?

patchie, 2021-06-14:
Hi,

I am new to Wasp-OS, and am not an Python expert in any way.
I have a win10 computer and a pine-time dev kit.

### Why am i creating an issue on the github page?
"Anyone can contribute to the wasp-os project." - ref: https://wasp-os.readthedocs.io/en/latest/contributing.html#introduction
"Wasp-os, the Watch Application System in Python, has one pervasive goal that influences almost everything about it, from its name to its development roadmap: make writing applications easy (and fun)." - ref: https://wasp-os.readthedocs.io/en/latest/appguide.html#introduction

Btw, i also tried contacting anyone related to wasp-os on the official IRC channel, which seems to be totally DEAD.

### What is the problem?
Well, its not quite intuitive to make an application for Wasp-OS at the moment.
I just want to highlight that we should try to update the tutorials, to make it more clear how to develop applications for the os. 
That is, if you want to make it "easy and fun" for everyone.

### What have i tried?
I tried to run the following code: https://wasp-os.readthedocs.io/en/latest/appguide.html#hello-world-for-wasp-os
I also watched the video here: https://www.youtube.com/watch?v=tuk9Nmr3Jo8
I soon found out i needed the wasp library, and tried to pip install it, but it seemed to not be the same..
After some time i found out i could download the whole wasp-os git code, and hack around, by moving files and such.
I think the solution for me was to just make a python file with the code under wasp-os-master\wasp, and then move all files from wasp-os-master\wasp\boards\simulator to the same folder.

I also had to move the simulator_skin.png to the following location: wasp-os-master\wasp\res\simulator_skin.png

To make it run on win, i also had to hack the wasp.py file as uname doesnt seem to work on win:
```
        try:
            if 'P8' in watch.os.uname().machine:
                self._nfylevels = [0, 225, 450]
            else:
                self._nfylevels = [0, 40, 80]
        except:
            print("not P8")
            self._nfylevels = [0, 40, 80]
```

I then also had to hack the code from the tutorial to be able to get any result:
```
# SPDX-License-Identifier: MY-LICENSE
# Copyright (C) YEAR(S), AUTHOR

import wasp

class HelloApp():
    """A hello world application for wasp-os."""
    NAME = "Hello"

    def __init__(self, msg="Hello, world!"):
        self.msg = msg
        self.foreground()   #added by patchie

    def foreground(self):
        self._draw()

    def _draw(self):
        draw = wasp.watch.drawable
        draw.fill()
        draw.string(self.msg, 0, 108, width=240)

HelloApp()   #added by patchie
```

The output:
```
= RESTART: C:\Users\.........\wasp-os-master\wasp\run.py
UserWarning: Using SDL2 binaries from pysdl2-dll 2.0.14

Warning (from warnings module):
  File "C:\Users\.........\wasp-os-master\wasp\display.py", line 77
    pixelview = sdl2.ext.pixels2d(windowsurface)
ExperimentalWarning: 'pixels2d' is in an experimental state.
MOTOR: set on
not P8
>>> 
```
And of course the watch, with "Hello, World!"

Is there any better ways of doing this?

Thanks in advance

- Patchie

Pittconnect, 2021-06-14:
Please join our PineTime Channel here https://discord.gg/b3AbYDg6 and currently have 287 members NOT all do PineTime and wasp-os BUT all do smartwatch programming currently 42 members online  my user name is Ra#8513 I have posted many links and I am building HRV app on PineTime using Wasp-os Let me know if you like to help !!

patchie, 2021-06-17:
### To run the simulator on Win10:
1. Make sure you have installed WSL2.
2. Install VScode
3. Install VScode extension called: "Remote - WSL"
4. Install MobaXterm, to be able to show linux gui on windows. Should work out of the box.
5. Download the git repository from WSL2 terminal window in VScode: git clone https://github.com/daniel-thompson/wasp-os
6. Open "wasp-os" directory and run "make sim"

daniel-thompson, 2021-06-17:
Thanks for the update. This looks a *much* cleaner approach to your original one... even to someone like me who hasn't used Windows for over a decade ;-) .

Are you interested into putting together a github PR to update the install guide?

pfeerick, 2021-06-24:
I'd wait a bit longer before doing any guides, as WSLg is just around the corner, and that should bring native GUI support to WSL2... bringing that list down to only five steps ;) I'm not sure why steps two or three are needed... since cloning and making the sim is all done from the WSL2 terminal.

patchie, 2021-06-24:
> I'd wait a bit longer before doing any guides, as WSLg is just around the corner, and that should bring native GUI support to WSL2... bringing that list down to only five steps ;) I'm not sure why steps two or three are needed... since cloning and making the sim is all done from the WSL2 terminal. 

Good point, but it will take some months, or half a year for it to be released, right?

Steps 2 and 3 is not needed, but it makes it a whole lot easier to develop in vscode. I can put it like this, if you dont know what vscode is, then you should probably use it. If you know what it is, you should still probably use it, but people tend to use the editor they are used to.

I have at lest never found a editor i could be comfortable with, before i found vscode. I have been looking for 10+ years...

And, for the people who wants to code in the terminal editor, be my guest, but it doesnt need to be in the tutorial. People who read tutorials and do as their told, doesnt want to use a termial editor.

pfeerick, 2021-06-24:
Yeah, I was waiting for today's Windows announcement to see what was happening, as it's been in the Insider builds since April, and says "WSLg is going to be generally available alongside the upcoming release of Windows". So It looks like it's coming in 21H2 (and hopefully still W10, or I'm probably sunk as my main PC doesn't have a TPM chip, let alone TPMv2 😆). 

So yes, looks like there's no point waiting as it's going to be a while. I would rephrase it, and re-order the steps to those that are basically 'required' and then 'optional' ... ie. 1,4,5,6 (using windows terminal/command prompt), and then 2,3 for easier code development.

I'm 100% with you on VSCode... it's my code editor of choice now. Before that, I used to use Notepad2, Notepad++, and Sublime Text - which I still use a lot, for for actual coding sessions VSCode usually gets spun up.