wasp-os/wasp-os #219: `.text' will not fit in region `FLASH_TEXT'

Pittconnect, 2021-06-14:
I get this error during   ` make -j `nproc` BOARD=pinetime all` The default make works fine but I am building an application HRV application that will display beats per minute or bpm, ibi or interval beat interval, and heart rate all at the same time on the pinetime display using wasp-os for which I have created a sample test data and its about 3.29MB and it seems like the file is to large for wasp-os to compile for pinetime is there a way to increase the file size so I do not get this Error  Here are the details on pastebin [https://pastebin.com/3d9RRtFi](url)

Pittconnect, 2021-06-15:
Seems like this is a known problem with Micropython when using other hardware https://github.com/micropython/micropython/issues/2527 So it not a hardware issue. Work Around is get the SDK upgrade for wasp-os SO Daniel when is the next upgrade to wasp-os coming, Now would be a good Time !

daniel-thompson, 2021-06-20:
The nRF52832 has only 512K of FLASH and about a third of it is reserved for the bluetooth stack. If you are running out of space then the one option is find things (from the manifests in the wasp/boards/ directory) to remove. A better option would be to store the large data files in the filesystem instead of embeddeding them in the main image.

> Seems like this is a known problem with Micropython when using other
> hardware micropython/micropython#2527 

I think this issue is for ESP26xx and ESP32 devices. I don't think there is any action needed for nRF52.