wasp-os/wasp-os #221: Accidentally uploaded reload-mcuboot.zip on Wasp-os (bricked?)

gandalf15, 2021-06-21:
Hello,

I would like to ask you for a help. I have sealed Pinetime. I was running the latest Wasp-os v0.4.1 on it. Then I wanted to try a new version of InfiniTime 1.2.0. I accidentally uploaded [reload-mcuboot.zip](https://github.com/JF002/InfiniTime/releases/download/0.14.1/reloader-mcuboot.zip) instead of the provided `reloader-factory.zip` that comes in wasp-os zip file. 
Now, the screen shows only smaller pine cone and blinks every ~ 3 sec (no BLE access). If I try to hold the button, I can see the wasp-os bootloader for a split second but it does not wait there and I do not have any BLE access. Is there any possibility to recover from this mess? If not, is it possible to open the watch and upload FW with cables?

Thank you for your replay.

daniel-thompson, 2021-06-22:
It is possible to recover without opening the device but it is a little tricky. In short you have to time the button release to occur just as the device is drawing the <- Start message and the boot logo.

The best way to do this is to get a feel for the boot loop cycle, hold the button and release it just as you *think* the device is about to reset (since that is also the time it will draw the Start message if it has been provoked into doing to).Then you have to rely a little on human error to get the timing exactly right. Eventually however it should end up showing the start message and be ready for a reflash.

I've forgotten exactly how I set up the bootloader but I think you have only a quarter second window to release the button which is why it works better to predict rather than to react.

gandalf15, 2021-06-22:
Thank you a lot for the detailed explanation. I managed to do it on the second try. Now, it is all working.

closed 2021-06-22 by gandalf15