wasp-os/wasp-os #224: Disable Bluetooth after boot?

inducer, 2021-06-23:
Suppose that
- I'm the security conscious type and not comfortable with random strangers being able to run Python on my watch (and modify state) without authentication (or set my watch, or snoop on my step count), or
- I'm on an airplane and they ask to turn off all transmitting devices, or
- I work in an EM-sensitive environment,

it'd be nice to have a knob in the UI to just turn off all use of Bluetooth.

I did some digging, getting as far as this:
https://github.com/micropython/micropython/blob/c99e4995fd6bb19fa4d0d587f1b72f46efcf9785/ports/nrf/modules/ble/help_sd.h#L37-L40
but all that running `ble.disable()` gets me is a reboot. :)

Is this realistic? If so, what's the most likely route?

Carglglz, 2021-06-23:
@inducer This is totally doable. 
in `micropython/ports/nrf/modules/ubluepy/modubluepy.c` add (in the `#if MICROPY_PY_BLE_NUS` block) above this line 
 https://github.com/daniel-thompson/micropython/blob/048bb2da83d0cbe73a829e461bd4f22b86c282bc/ports/nrf/modules/ubluepy/modubluepy.c#L56
```
STATIC mp_obj_t ubluepy_uart_advertise(void)
{
	ble_uart_advertise();
	return mp_const_none;
}
STATIC MP_DEFINE_CONST_FUN_OBJ_0(ubluepy_uart_advertise_obj,
		                 ubluepy_uart_advertise);

STATIC mp_obj_t ubluepy_uart_advertise_stop(void)
{
	ble_drv_advertise_stop();
	return mp_const_none;
}
STATIC MP_DEFINE_CONST_FUN_OBJ_0(ubluepy_uart_advertise_stop_obj,
		                 ubluepy_uart_advertise_stop);
```
And below this line https://github.com/daniel-thompson/micropython/blob/048bb2da83d0cbe73a829e461bd4f22b86c282bc/ports/nrf/modules/ubluepy/modubluepy.c#L80

```
{ MP_ROM_QSTR(MP_QSTR_uart_advertise),  MP_ROM_PTR(&ubluepy_uart_advertise_obj) },
{ MP_ROM_QSTR(MP_QSTR_uart_advertise_stop), MP_ROM_PTR(&ubluepy_uart_advertise_stop_obj) },
```
Then in MicroPython you can import these functions to enable or disable bluetooth advertising.

e.g. I've tested it adding these two functions in `watch.py` 

```
from ubluepy import uart_advertise, uart_advertise_stop
```

Then they are available in `wasp.py` as `watch.uart_advertise()`, and `watch.uart_advertise_stop()`.
From there you can add any logic you want that meets your needs 👍 . e.g. I linked advertising to the system/screen wake/sleep so it only advertises as long as the screen is on. This probably saves some battery life too. #215

fgaz, 2021-09-28:
> but all that running ble.disable() gets me is a reboot. :)

I experimented a bit and found that the reboot is caused by the console connection getting interrupted, not by disabling the bluetooth. This means that disabling bluetooth from the watch works. Here's a small app to do that:

```python
import wasp
import ble

class DisaBLEApp():
    NAME = 'DisaBLE'

    def foreground(self):
        draw = wasp.watch.drawable
        draw.fill()
        ble.disable()
        draw.string('Disabled!', 0, 60, width=240)
```

Unfortunately re-enabling it with the same method does not work:

```
OSError: Can not notify attribute value. status: 0x02
```

thiswillbeyourgithub, 2022-02-28:
@fgaz I took the liberty of integrating it with the settings app. Is that okay?