wasp-os/wasp-os #226: Unable to download data using pull command on sealed PineTime using wasptool

Pittconnect, 2021-06-29:
[image]
What I have tried I have tried flashing reloader_mcuboot.zip and micropython.zip for the sealed device.

Pittconnect, 2021-07-10:
[image]

daniel-thompson, 2021-07-15:
Please do not post pictures of text. Post the actual text instead. Similar I don't understand the picture of the spreadsheet. Please post the actual underlying data and explain what you think it wrong with it.

PS to delete files from the device you can use --eval and issue the appropriate python expression for that.

GitZero1, 2021-07-16:
Hi Daniel I have been trying to help with this issue so maybe I can be more clear on the problem he is having. In short he is trying to follow these steps https://wasp-os.readthedocs.io/en/latest/apps.html?highlight=hrs.data#heart-rate-monitor 

In his screenshot he has done it incorrectly but since then, both myself and Pitconnect have done the following

1) navigate to heartrate app on watch
2) execute --eval debugging command
3) wait ~10 min with heart app running
4) use --pull hrs.data

This results in OSerror: 2

I have also treid 
after step 3 close the heart app THEN pull
after step 3 run --eval to disable debugging, then pull
after step 3 run --eval to disable debugging, close the heart app, then pull

All of this just results in OSerror: 2 when the pull commnad is issued. 

I'm not sure what is being misunderstood here as I found your instructions very clear yet I was unable to make it work. A nudge in the right direction would be greatly appreciated, thanks!

daniel-thompson, 2021-07-16:
Works for me (see below).

Can you be a bit more liberal with details? For starters always share logs of the command errors (as text... never as pictures) and, similarly, always provide version information with any bug report (this is a very new feature, the most obvious reason for this not to work is because the firmware is too old).

~~~
<Select the HR app>
stingray$ ./tools/wasptool --eval 'wasp.system.app.debug = True'
<Wait for the BPM value to update (twice)>
stingray$ ./tools/wasptool --pull hrs.data
Downloading hrs.data:
[##################################################] 100%       
stingray$ # Close the HR app
stingray$ ./tools/wasptool --eval 'import os; os.remove("hrs.data")'
<Resetting is less effort than clearing the debug flag on the HR app>
stingray$ ./tools/wasptool --reset
~~~

GitZero1, 2021-07-16:
OK I feel I should apologize! I was not aware this was a new feature and was using firmware I had built some time ago. I cloned a fresh copy of your repo, built and followed the instructions with no issue. I will attempt to help PitConnect when I see him online again. Thanks Daniel!

daniel-thompson, 2021-07-16:
No need to apologize! Having a community of people helping each other is always a delight.

Pittconnect, 2021-07-17:
GitZero1 has helped he has different setup than I do so we compared and notes. And it turns out that I was using Android DaFtlasher to flash the firmware on the Pinetime that some how violates a security protocol of google IS what I am guessing SO I switch to ./tools/ota-dfu/dfu.py to upload the firmware and that has helped. So the new firmware build and uploading the firmware using dfu.py seems to fix the problem I have not tried to duplicate the problem using daflasher and android 5.1  You may mark this as solved and close the Issue Daniel Thanks

daniel-thompson, 2021-07-17:
Glad you have solved the problem.

Note that the wasp-os installation instructions recommend using nRF Connect to install from Android devices. This is not because DaFlasher doesn't work (it works just fine and is *way* faster than nRF Connect). However on my phone DaFlasher was also more likely to "randomly disconnect" than nRF Connect. Disconnecting is harmless... just send it again. However since it is disconcerting for users so we recommend nRF Connect instead in the install docs. It does cross my mind that Android 5.1 devices are usually rather old. It is possible you device doesn't support Bluetooth Low Energy at all.

closed 2021-07-17 by daniel-thompson