wasp-os/wasp-os #227: upload bootloader into infinitime-pinetime with wasptool?

wilsonjwcsu, 2021-07-01:
How can we go about installing the bootloader from a linux command line, starting with a PineTime that has the infinitime bootloader? Here's what I tried, but it's complaining that the watch isn't in DFU mode. If I restart the watch, I get the infinitime bootloader, not the wasp-os bootloader with bluetooth icon.

```
user@machine:~/git/wasp-os$ tools/ota-dfu/dfu.py -z ./build-pinetime/reloader-mcuboot.zip -a E1:1B:6D:8F:EB:4D --legacy
Sending file reloader-mcuboot.bin to E1:1B:6D:8F:EB:4D
Binary imge size: 304556
Binary CRC32: 3987761949
Connecting to E1:1B:6D:8F:EB:4D
Need to switch to DFU mode
Exception at line 154: UUID not found: 00001531-1212-efde-1523-785feabcd123
```

Pittconnect, 2021-08-22:
@wilsonjwcsu  hello Jessee, what version of infinite time had you installed.  If the system has a 1.0 or a newer InfiniteTime bootloader it will not downgrade to wasp-os bootloader 0.4 As InfiniteTime 1.0 came after wasp-os 0.4 But it may allow you to downgrade from InfiniteTime 1.0 bootloader to InfiniteTime 0.98 bootloader. When you first run the 
`make -j `nproc` BOARD=pinetime all `
command Notice it create not only reloader_mcuboot.zip but also reloader-factory.zip this is the factory reloader created prior to Infinitetime 1.0 so it will allow you to load wasp-os reloader-mcuboot.zip  I hope this helps !!

Siroj42, 2022-01-24:
I just got my new PineTime, and Siglo worked. You can set a custom file to upload in the app.