wasp-os/wasp-os #229: Profile build of wasp-bootloader

shermozle, 2021-07-02:
Installation doc says:

> Copy reloader-mcuboot.zip (see Building wasp-os from source) to your Android device and download nRF Connect for Android if you do not already have it.

To enable a quick and easy try out of the OS, it'd be great to include a built version of this. Otherwise you have to go through the quite onerous process of installing all the prerequisites (the ARM GCC bits are particularly involved) and compiling it all.

daniel-thompson, 2021-07-02:
Scroll down ;-). The section immediately after the instructions on build from source describe exactly this: https://wasp-os.readthedocs.io/en/latest/install.html#binary-downloads

shermozle, 2021-07-02:
Where is reloader-mcuboot.zip?

Itai-Nelken, 2021-07-02:
In the tar.gz archive in the release

closed 2021-07-15 by daniel-thompson