wasp-os/wasp-os #230: wasp.watch.drawable.wrap() breaks short lines containing spaces

fgaz, 2021-07-18:
```
>>> wasp.watch.drawable.wrap("a b",240)
[0, 2, 3]
```

I'd expect it to return `[0, 3]` (like `wrap("abc")`), since it will fit in 240 pixels

Pittconnect, 2021-07-19:
>>> wasp.watch.drawable.wrap("a,0",240)
[0, 3]
you need to put a place holder to get [0, 3] as output for your wrap command

fgaz, 2021-07-19:
@Pittconnect with the current behavior yes, but why should "a b" be split if it doesn't exceed 240 characters? Word wrapping usually doesn't turn _every_ space into a newline

> you need to put a place holder to get [0, 3] as output for your wrap command

 is there a placeholder that is guaranteed to have the same width as a space?

Pittconnect, 2021-07-20:
I am not sure what your questions is ? Or what kind of solution you are looking for ? I just responded to your post base on what you posted was `I'd expect it to return [0, 3]` If you wish to engage in intellectual debate please post your request in a group https://t.me/pinetime, The github issue is to track bugs you find related to wasp-os usage This is not a bug, You split ab into a b which mean ab is acquired 2 space while a b has acquired 3 space  Here are some more examples 
>>> print(watch.rtc.get_localtime())
(2020, 3, 1, 14, 14, 38, 6, 61)
>>> print("WJS""GPS")
WJSGPS
>>> wasp.watch.drawable.wrap("cd",240)
[0, 2]
>>> wasp.watch.drawable.wrap("c d",240)
[0, 2, 3]
>>> wasp.watch.drawable.wrap("a   b",240)
[0, 4, 5]
>>> wasp.watch.drawable.wrap("a    b",240)
[0, 5, 6]
>>> wasp.watch.drawable.wrap("e",240)
[0, 1]
>>>

daniel-thompson, 2021-07-20:
> The github issue is to track bugs you find related to wasp-os usage. This is not a bug.

I think this is a bug... or, the very least, seriously counter-intuitive.

The wasp-os wrap() function is used to determine where strings should be broken during multi-line rendering. "a b" easily fits onto a single line but wrap() is placing a line break between the two "words" (in other words it is returning `[0, 2, 3]` when `[0, 3]` seems to be more correct).

The effect also affects other strings such as "a b c" which also fits onto a single line but, once again, wrap() is generating a line break between the final two "words" of the string:
```
>>> # W is a very wide character (usually wider than a space)
>>> wasp.watch.drawable.wrap('aWb', 240)
[0, 3]
>>> wasp.watch.drawable.wrap('a b', 240)
[0, 2, 3]
>>> wasp.watch.drawable.wrap('a b c', 240)
[0, 4, 5]
>>> wasp.watch.drawable.wrap('a b c d', 240)
[0, 6, 7]
```

daniel-thompson, 2021-07-22:
Should now be fixed. Please try with most recent master branch.

closed 2021-07-22 by daniel-thompson

fgaz, 2021-07-24:
Works great, thanks!

commit 0831f79a (2021-07-22): draw565: Improve line wrapping

Currently the final word of wrapped text will always appear as a single
word on its own line. Fix this by rearranging the break cases to avoid
searching for the most recent space when we get to the end of the text.

Fixes: #230